ARMmbed/mbed-os #7256: ATParser.cpp: incompatible redefinition of macro MIN

teetak01, 2018-06-19:
### Description
Mbed OS 5.9.1
Component: ARM Compiler 5.06 update 3 (build 300)

```
ATParser.cpp@37,0:  #47-D: incompatible redefinition of macro "MIN" (declared at line 145 of "./mbed-os/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/drivers/fsl_common.h")
```
[ ] Question  
[ ] Enhancement  
[X] Bug

jarvte, 2018-06-21:
There is no ATParser.cpp file anymore in mbed-os. Renaming of ATParser.cpp to ATCmdParser was done in May 2017. ATCmdParser in mbed-os 5.9.1 does not declare MIN.
Could you please clarify the issue?

teetak01, 2018-06-21:
It is possible one of the drivers inside easy-connect is still pulling in some old seperate implementation of ATParser.cpp

0xc0170, 2018-06-21:
> It is possible one of the drivers inside easy-connect is still pulling in some old seperate implementation of ATParser.cpp

Seems like this is not an issue for Mbed OS but rather a component?

jarvte, 2018-06-21:
@0xc0170 I agree, nothing to fix in mbed-os

teetak01, 2018-06-21:
The file is actually inside wifi-ism43362, I will close this.

closed 2018-06-21 by teetak01