ARMmbed/mbed-os #7257: REALTEK_RTL8195AM defines DEVICE_EMAC flag for WiFi driver

SeppoTakalo, 2018-06-19:
### Description

As explained in porting sections, device should provide `EMAC::get_default_instance()` only if pure Ethernet. Please see:
https://os.mbed.com/docs/v5.9/reference/ethernet-port.html

Also, device should enable `EMAC` flag in `targets.json` only if its pure Ethernet. Please see:
https://github.com/ARMmbed/mbed-os/tree/master/features/netsocket/emac-drivers#device_emac

REALTEK_RTL8195AM does both for WiFi interface. It provides `EMAC::get_default_instance()` and also enables `EMAC` flag in the `device_has` section. Both should be removed and https://github.com/ARMmbed/mbed-os/issues/7246 should be fixed by providing default WiFi interface, like this:

```
#include "RTWInterface.h"

WiFiInterface *WiFiInterface::get_target_default_instance()
{
    static RTWInterface wifi;
    return &wifi;
}
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-06-19:
@ARMmbed/team-realtek Please review

samchuarm, 2018-07-02:
@M-ichae-l Hi Michael, please review and address this for RTL8195AM EMAC implementation. Thanks.

M-ichae-l, 2018-07-02:
@samchuarm  OK. Thank you.

M-ichae-l, 2018-07-03:
I have verified the #7259. I will PR (updating the rtl8195am lib and remove the DEVICE_EMAC) soon.

samchuarm, 2018-07-09:
Thanks @M-ichae-l  Please ref the PR here once you submit it. Thx.

samchuarm, 2018-07-19:
Hi @M-ichae-l , any update on the PR?

M-ichae-l, 2018-07-19:
@samchuarm 
I will PR "remove the DEVICE_EMAC" today. The rtl8195am lib update will together with low power feature in future.

M-ichae-l, 2018-07-19:
Please refer to #7551 for the fix.

SeppoTakalo, 2018-07-23:
Once the #7551 is merged, then I think this can be closed.

samchuarm, 2018-07-26:
Hi @SeppoTakalo , it seems #7551 has been merged now. Thanks.

PR #7259 (2018-06-20): Provide default WiFi interfaces for Ublox and Realtek
### Description

To allow application to choose default WiFi interface by calling `WiFiInterface::get_default_instance()` provide its implementation on these two targets.

Partially fixes:
#7257 
#7246

### Pull request type
    [X] Fix
    [ ] Refactor
    [ ] New target
    [ ] Feature
    [ ] Breaking change
closed 2018-07-27 by 0xc0170