ARMmbed/mbed-os #7278: nRF52832 SPI Driver Failure

AGlass0fMilk, 2018-06-20:
### Description

Toolchain: GCC ARM
Target: nRF52832 (custom board)

I recently synced my fork to mbed-os 5.9. I was previously at this commit hash:
16ece66f950facea0cc06047794afdd0eab39372

Now when I run the same code, I get the following error related to the nRF SPI driver:
```
++ MbedOS Error Info ++
Error Status: 0x80ff0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x489af
Error Value: 0x0
Current Thread: Id: 0x20005af4 Entry: 0x4a815 StackSize: 0x1000 StackMem: 0x20005b40 SP: 0x200069f8
-- MbedOS Error Info --
nrf failure at /home/gdbeckstein/HeadsUp/hud-devices/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/drivers_nrf/spi▒▒d> ɉTj
```
Just to make sure it wasn't my code, I checked out my fork prior to the sync and recompiled everything. No issue. Not sure what changes have occurred that may cause this problem

@marcuschangarm Any idea? For reference, I'm using a custom pinmap that allocates two C++ SPI instances to hardware instance 2. They have similar pinouts but talk to different chips with different clock lines. Excerpt from my pinmap override:

```
/** Assign both the SPI bus pins to hardware instance 2 */
const PinMapSPI PinMap_SPI[] = {
		{ SPI_MOSI_PIN, 	NC, 				SPI_CLOCK_PIN, 		2 },
		{ SPI_MOSI_PIN, 	SPI_MISO_PIN, 	SPI_CLOCK_ALT_PIN, 	2 },
		{ NC, 				NC, 				NC, 						NC }
};
```

I'll continue to debug from my side.

### Issue request type

[ ] Question  
[ ] Enhancement  
[ X ] Bug

SenRamakri, 2018-06-20:
@AGlass0fMilk - Are you running release build? If not you must have gotten more terminal output. If you are running release build, it may be useful to run in develop/debug profile to capture more info on this error.

AGlass0fMilk, 2018-06-20:
@SenRamakri I am running with the debug profile. Are there any additional debug flag options I can enable?

SenRamakri, 2018-06-20:
@AGlass0fMilk - Ok, Thanks for confirming. Is the statement "nrf failure at /home/gdbeckstein/HeadsUp/hud-devices/mbed-os/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET..." coming out in terminal or something you added to the description above? If that's part of terminal output its working as expected.

SenRamakri, 2018-06-20:
@AGlass0fMilk - Partly error reporting and to identify where the actual problem is. Looks like reporting is working as expected providing you enough info on the error scenario.

marcuschangarm, 2018-06-20:
@AGlass0fMilk Did you mean 5.9 or 5.9.1? The commit you mentioned working is between the two releases.

AGlass0fMilk, 2018-06-20:
I periodically pull to the most recent commit on master. I thought I was updating to 5.9 but maybe the blog post announcing 5.9 came a bit after when I synced with that commit. So I must've meant 5.9.1

marcuschangarm, 2018-06-20:
@AGlass0fMilk 

I think I found the problem. Can you give this a spin please? https://github.com/ARMmbed/mbed-os/pull/7280

AGlass0fMilk, 2018-06-21:
@marcuschangarm I just synced with master. The crash issue is gone. However, I have an eventqueue running a sensor polling system and it seems to be sporadically being executed now. It used to poll sensors every few sensors. Now the debug UART shows output at random, sometimes dumping several samples all at once.

AGlass0fMilk, 2018-06-21:
Scratch the slowdown issue for now. This has been resolved by PR #7280.

Thanks @marcuschangarm !

closed 2018-06-21 by AGlass0fMilk

marcuschangarm, 2018-06-22:
Awesome! Does that mean this issue is resolved as well? https://github.com/ARMmbed/mbed-os/issues/6623