ARMmbed/mbed-os #7281: Project file exported to IAR does not enabled multi threaded c lib

Alex-EEE, 2018-06-20:
mbed os 5.8.6

When exporting to IAR, the project IS NOT configured to use the multithreaded C / C++ lib!!

Exporter should enable that option.

    [] Question
    [ ] Enhancement 
    [ X ] Bug

0xc0170, 2018-06-21:
Hi @Alex-EEE ,

`threaded_lib` is defined in the linker options (same effect as the IDE selection)

Taken from IAR release notes for 7.70

```
Configuration of the runtime library for use with threads (ref EW23993)
To configure the runtime library for use with threaded applications, use the linker option --threaded_lib, alternatively select
Project>Options>General Options>Library Configuration>Enable thread support in library
Read more about Managing a multithreaded environment in the C/C++ Development Guide.
```

0xc0170, 2018-06-21:
From the linker extra options:

```
--skip_dynamic_initialization
--threaded_lib
```

closed 2018-06-21 by 0xc0170

Alex-EEE, 2018-06-21:
Awesome, thanks for the quick response!