ARMmbed/mbed-os #7282: STM32L4+ Support Roadmap?

AGlass0fMilk, 2018-06-21:
### Description
I am looking at using the STM32L4+ series of chips and was going to buy the  NUCLEO-L4R5ZI development kit to get started. I noticed mbed does not yet support STM32L4+ chips. Is there any roadmap of when mbed will support these platforms? Or would it be simple to adapt from existing ports?

### Issue request type

[ X ] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-06-21:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-06-21:
Hi,

These devices are not planned to be done by ST team for the moment.

If someone wants to do the porting, you can refer to this wiki page https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform for porting information.

For the PeripheralPins.c file creation, you can use `mbed-os\tools\targets\STM32_gen_PeripheralPins.py` python tool (but you have to install ST CubeMx software first).

We'll review the code and make full non-reg tests on all toolchains. If all is OK this board will be published on mbed.

Thanks.

AGlass0fMilk, 2018-06-21:
Thanks, @bcostm. That answers my question.

closed 2018-06-21 by AGlass0fMilk