ARMmbed/mbed-os #7292: RFC_864_pattern is not correct with xinetd TCP server

jeromecoutant, 2018-06-21:
### Description

I set up a local TCP server in order to execute netsocket tests.

I am using xinetd, and configured services as recommened:

discard     stream  tcp6    nowait  root    internal
discard     dgram   udp6    wait    root    internal
echo        stream  tcp6    nowait  root    internal
echo        dgram   udp6    wait    root    internal
chargen     stream  tcp6    nowait  root    internal
chargen     dgram   udp6    wait    root    internal
daytime     stream  tcp6    nowait  root    internal
time        stream  tcp6    nowait  root    internal

Then, there are some issues with the chargen service in TCPSOCKET_RECV_100K sub test:
https://github.com/ARMmbed/mbed-os/blob/443e37b00c6882756bdeecfd41f45cdd2efaf872/TESTS/netsocket/tcp/tcpsocket_recv_100k.cpp#L58

After several checks, it seems there are several misalignments with what inetd service is providing:

- Pattern starts with !, not space
 
- 2 characters are missing : ~ and space

- sometimes, 2 lines are repeated : (maybe every 94*74 characters)

```
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefgh
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghi
!"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefgh
"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghi
```

I think current xinetd code is wrong…
https://github.com/xinetd-org/xinetd/blob/e7c1ba41f4f86b436fb82b0d55cd5d387bd4ecc4/xinetd/builtins.c#L377
 
But mbed-os test has to be updated with the current situation...

### Issue request type

[ ] Question  
[x] Enhancement  
[ ] Bug

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/ONME-3733*

SeppoTakalo, 2018-11-14:
Fix has been submitted in #8645 so this can be closed.

closed 2018-11-14 by jeromecoutant