ARMmbed/mbed-os #7293: Call greentea_metrics_report even in case of TIMEOUT

jeromecoutant, 2018-06-21:
### Description

At the end of each greentea test, some information are sent to the console.
https://github.com/ARMmbed/mbed-os/blob/443e37b00c6882756bdeecfd41f45cdd2efaf872/features/frameworks/greentea-client/source/greentea_metrics.cpp#L61

This is good, but only available with OK or FAIL status, not with the TIMEOUT.

Do you think it is possible to get the same console information even after timeout has expired?

Thx

### Issue request type

[ ] Question  
[x] Enhancement  
[ ] Bug

aashishc1988, 2018-08-15:
Possible reasons for TIMEOUT:
1. Not being able to sync with the device for some reason
2. Broken UART
3. Flaky device (connects/disconnects on whim, possibly due to faulty uart)
4. Test itself had an issue/bug which resulted in device being unresponsive

greentea_metrics_resport is used to send heap, stack and cpu info. In cases 1, 2, 3, I dont see any point of generating that information as we would get nothing useful out of it. In case 4, if the device itself became unresponsive, I am not sure how we can generate that information anyway. 

@jeromecoutant Can you please define a use case where generating metrics report would be useful?

cc: @studavekar

jeromecoutant, 2018-08-16:
Other reasons for TIMEOUT:
- device want to sleep and is not able to wake up for some reason
- device ticker implementation is wrong
- device went to a wrong internal state
- ...

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-171*

aashishc1988, 2018-10-05:
@jeromecoutant Sorry for late reply, I got caught up on other things and completely missed this.
Looking at the test architecture, I failed to understand a use case where this would be needed or actually work. So, right now, if the timeout happens due to any of those reasons indicated above, the final step in the host test, that is, publishing test results, 
`GREENTEA_TESTSUITE_RESULT`
doesn't get executed. Because in most of those cases we are stuck in while(1) (caused due to mbed_error, and we do print meaningful info there about why and where it happened) and eventually htrun sees we have surpassed the test time limit and we exit out of the test. Therefore, we have no way of executing that part of the code where we publish testsuite results. 
So, right now, with the current architecture, this request is not possible. However, we can move publishing of test results part more upstream, that is, calling it in the context of greentea than the individual test, but that sort of change requires a very strong rationale. `greentea_metrics_report` reports heap and stack information, if the test failed to execute and timed out, and we are publishing stack and error information via mbed_error, what other use case there be where we would need to execute `greentea_metrics_report`

jeromecoutant, 2018-10-08:
Thx @aashishc1988 t analyse any possible enhancement.
My global request is to add debug information around sleep and deepsleep mode.

aashishc1988, 2018-10-17:
@jeromecoutant can you please open a new issue explaining the problem in bit more detail and what exactly is required? I am going to close this issue.

jeromecoutant, 2018-10-19:
I don't understand why I need to open a new issue in order to close 1 issue...

closed 2018-10-19 by jeromecoutant

aashishc1988, 2018-10-19:
@jeromecoutant The new request is quite different to the original intent of this issue and discussion we had. In order to avoid confusion, and keep this discussion in records so that we know why the current request was not possible, it is better to open a new request. Sorry for the inconvenience.