ARMmbed/mbed-os #7297: nRF52 BLE VS UUID Crash

AGlass0fMilk, 2018-06-21:
### Description
I recently synced my mbed-os fork and now I am experiencing a crash associated with BLE initialization.
My BLE profile requires me to exceed the default vendor-specific UUID limit of 4. Before I updated, this limit was set by a macro in the TARGET_NORDIC/TARGET_NRF5x/TARGET_SDK_14_2/TARGET_SOFTDEVICE_S132/mbed_lib.json file.

Now the system has been changed to be overwritable by a user's mbed_app.json file. The new limit is set by a configurable macro in features/FEATURE_BLE/targets/TARGET_NORDIC/mbed_lib.json.

I have configured this macro to allow 8 Vendor specific UUIDs (what I had set it to under the previous system). Now, when BLE is initialized, my chip resets with no error information from mbed -- this leads me to believe it is the soft device causing a reset (I am in debug profile).

The strange thing is, if I configure the macro to have less than 8 VS UUIDs (I tried 7), initialization works as expected but of course the VS UUID services/characteristics aren't shown after a 7 have been added. If I go above 8 (I tried 10 and 12) then the chip doesn't reset but it is **not visibly broadcasting** (I have checked using the Punch Through BLE app and nRF Connect desktop).

So I'm not sure what's going on or what has changed with this configuration system. The most recent stable commit I was building on was 16ece66f950facea0cc06047794afdd0eab39372.

### Issue request type

[ ] Question  
[ ] Enhancement  
[ X ] Bug

AGlass0fMilk, 2018-06-22:
Crashing was occuring when nordic-ble.uuid_table_max_entries was set to 8 and the NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE was set to the default of 0x600. I then tried increasing the gatt attribute table size to 0x800. Then I started getting a fatal crash that seemed to be due to main stack corruption. 

I figured this was due to the soft device using more RAM than previously so I adjusted the allocation in the linker script which stopped the crashing but the device was still not broadcasting.

I'll do some more debugging tomorrow. I may just switch all my custom services/characteristics to share the same base VS UUID to avoid this problem altogether...

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-215*

closed 2018-11-05 by ciarmcom