ARMmbed/mbed-os #7300: NRF52 BLE persistent bonding

andrewleech, 2018-06-22:
### Description
I'm using NRF52 on recent master (currently from 94a06cef) with the new SecurityManager:
https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/connectivity/bluetooth/SecurityManager.md

Thanks so much, it's fantastically easy to use the get secure le pairing working, far easier than trying to find (non-existent?) nordic docs/examples on how to achieve the same.

However, am I right in thinking there is no on-chip persistence available currently for bonding information on the nrf52?

The SecurityManager appears to require the usage of mbed filesystem, which looks like it will need off-chip storage?
It also appears that nordic's implementation of persistence on-chip has been left out of the current 14.2 implementation in the codebase?

Is there any way to use bluetooth bonding on the nrf52 currently that doesn't require either a separate flash chip or re-pairing the phone after ever reboot of the nrf52?

### Issue request type

    [X] Question
    [ ] Enhancement 
    [ ] Bug

pingdan32, 2018-06-22:
Precisely, I am currently using BLE at the same time need to write some data to the flash, so that the device is used when restarting. After testing, it has been found that writing FLASH often fails when there is a BLE connection, or the data read when restarting is incorrect. What should I do to deal with this situation?

andrewleech, 2018-06-22:
Another project at work got flash writing with ble active, on nrf52 with the previous nrf SDK in mbed. I know they had significant issues, including I think some bugs relating to interrupts. I'll chat to the guys involved with that on Monday for some pointers.

marcuschangarm, 2018-06-22:
The FlashIAP implementation should be writing to the flash in collaboration with the SoftDevice and not cause disconnections. 

However, the FlashIAP calls can return with a failure error code to indicate the SoftDevice is busy and the that the user should retry the operation. 

@pingdan32 Do you currently perform a retry when the write operations return a failure error code?

@pan- Have you noticed disconnections when using FlashIAP on the NRF52?

pan-, 2018-06-22:
@marcuschangarm This was out of scope for this release so it is uncharted territory for us.

marcuschangarm, 2018-06-22:
@pan- Thank you for the clarification!

pingdan32, 2018-06-23:
@marcuschangarm Yes, I do retry it. However, sometimes there will be the following situations: 
step 1,  Ensure that an BLE connection is established as Peripheral role. 
step 2, Call `set` method and confirm that it has returned NVSTORE_SUCCESS; 
step 3, Call `get` method. But it returns NVSTORE_DATA_CORRUPT. 
Why? I have no way to do this.

pingdan32, 2018-06-23:
I'm using NVStore component.

pingdan32, 2018-06-26:
I insert a debug statement as following:
```
for (int i = 0; i < num_write_retries; i++) {
        ret = _flash->program(buf, _flash_area_params[area].address + offset, size);
        printf("%s: [%d] program flash result: %d\r\n", __FUNCTION__, i, ret);
        if (!ret) {
            return ret;
        }
        wait_ms(1);
    }
    return ret;
```

and my code is as following ( Ensure that an BLE connection is established as Peripheral role):

```int weight = 10;
int res = NVStore::get_instance().set(1, sizeof(int), &weight);
if (res != NVSTORE_SUCCESS) {
      printf("FLASH: store weight failed: %d\r\n", res);
} else {
      printf("FLASH: store weight success\r\n");

      uint16_t actualSize = 0;
      weight = 0;
      res = NVStore::get_instance().get(1, sizeof(int), &weight, actualSize);
      if (res != NVSTORE_SUCCESS) {
            printf("FLASH: load weight failed: res=%d\r\n", res);
      } else {
            printf("FLASH: load weight success: weight=%d\r\n", weight);
      }
}
```

but the log is as following:

```
flash_write_area: [0] program flash result: 0
flash_write_area: [0] program flash result: 0
FLASH: store weight success
FLASH: load weight failed: res=-4
```

marcuschangarm, 2018-06-26:
Thank you for the trace! It rules out my initial suspicion.

And you can confirm it only happens when BLE is connected and never when it is not?

pingdan32, 2018-06-27:
Yes, i sure. 
```
int main() {
    int weight = 18;
    int res = NVStore::get_instance().set(1, sizeof(int), &weight);
    if (res != NVSTORE_SUCCESS) {
        printf("FLASH: store weight failed: %d\r\n", res);
    } else {
        printf("FLASH: store weight success\r\n");
        uint16_t actualSize = 0;
        weight = 0;
        res = NVStore::get_instance().get(1, sizeof(int), &weight, actualSize);
        if (res != NVSTORE_SUCCESS) {
           printf("FLASH: load weight failed: res=%d\r\n", res);
        } else {
           printf("FLASH: load weight success: weight=%d\r\n", weight);
        }
    }

    while (true) {
        led = !led;
        wait(0.5);
    }
}
```

The log is as following:
```
flash_write_area: [0] program flash result: 0
flash_write_area: [0] program flash result: 0
flash_write_area: [0] program flash result: 0
flash_write_area: [0] program flash result: 0
FLASH: store weight success
FLASH: load weight success: weight=18
```

marcuschangarm, 2018-06-27:
Interesting. I'll have to give it some thought. 

Thank you for the help!

marcuschangarm, 2018-07-03:
@tanoc Thank you for the explanation. I see now that the code is taking different paths if the SoftDevice is enabled.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPAN-214*

closed 2018-11-05 by ciarmcom

bsturgess, 2019-05-17:
Hello, Seeing as its not possible to use FlashIAP to write to the flash when BLE is enabled. is there a way to temporarily shut shown BLE to enable the ability to write?