ARMmbed/mbed-os #7301: Redefined macros

leibin2014, 2018-06-22:
### Description
Dear experts,
There are two header files mbed_rtx_conf.h and RTX_Config.h with a lot of duplicated defines such as `OS_STACK_SIZE`, `MBED_CONF_APP_THREAD_STACK_SIZE`. And different values defined for the same macro in different files. Even if there is some `#ifndef` checking before `define`, we still can see a lot of compile warnings when we compile the code.
Further more, since different values are defined for the same macro in different header files, then could have risk that different source files include different header file will use different default values. Can this cause conflicts?
So I'm asking do we plan to merge these two header files into single to fix the compile warnings and avoid risks?

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-06-22:
> So I'm asking do we plan to merge these two header files into single to fix the compile warnings and avoid risks?

No, because:

RTX_Config provides default values that are set inside RTX. In case a user do not set those, use default. This file is provided with RTX. Do not change this file!

mbed_rtx_conf.h - user settings - in this case it's Mbed OS configuration for RTX.

I would like to know what warnings are you getting? Can you paste them here, they should be fixed.

leibin2014, 2018-06-25:
@0xc0170 The warnings as below:
```compiling rtx_msgqueue.c
compiling mbed_mktime.c
compiling rtx_evflags.c
compiling rtx_kernel.c
compiling ConditionVariable.cpp
compiling EventFlags.cpp
compiling rtx_system.c
compiling rtx_mutex.c
compiling Semaphore.cpp
compiling TableCRC.cpp
compiling rt_OsEventObserver.c
compiling equeue_posix.c
compiling mbed_tz_context.c
compiling rtx_thread.c
compiling MbedCRC.cpp
compiling RTX_Config.c
compiling rtx_semaphore.c
compiling rtx_mempool.c
compiling equeue.c
compiling rtx_timer.c
compiling rtx_evr.c
compiling Mutex.cpp
compiling cmsis_os1.c
C:/MeshEmbedded-0.0.9/tools/toolchain/bin/arm-none-eabi-g++.exe -g -Wall -Wextra -pipe "-fvisibility=default" -x c "-std=gnu99" -g3 -fomit-frame-pointer -fno-exceptions -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -fno-delete-null-pointer-checks -ffunction-sections "-fmessage-length=0" -fdata-sections -fsigned-char -mthumb -MMD "-mcpu=cortex-m4" "-mfloat-abi=softfp" "-mfpu=fpv4-sp-d16" -D__CORTEX_M4 -DARM_MATH_CM4 "-DMBED_BUILD_TIMESTAMP=1528862553.02" "-DDEVICE_CRC=1" "-DDEVICE_I2C=1" "-D__MBED__=1" "-DDEVICE_I2CSLAVE=1" "-D__FPU_PRESENT=1" "-DDEVICE_PORTOUT=1" -DUSBHOST_OTHER "-DDEVICE_PORTINOUT=1" -DTARGET_RTOS_M4_M7 "-DDEVICE_LOWPOWERTIMER=1" "-DDEVICE_RTC=1" -DTOOLCHAIN_object "-DDEVICE_SERIAL_ASYNCH=1" -D__CMSIS_RTOS "-DDEVICE_ANALOGOUT=1" -DTOOLCHAIN_GCC "-DDEVICE_CAN=1" -DTARGET_STM32L486xG -DTARGET_CORTEX_M "-DDEVICE_I2C_ASYNCH=1" -DTARGET_LIKE_CORTEX_M4 -DTARGET_STM32L486RG -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_STM32L4 "-DDEVICE_LPTICKER=1" "-DDEVICE_SPI_ASYNCH=1" "-DDEVICE_PWMOUT=1" "-DDEVICE_INTERRUPTIN=1" -DTARGET_CORTEX "-DTRANSACTION_QUEUE_SIZE_SPI=2" -D__CORTEX_M4 "-DDEVICE_STDIO_MESSAGES=1" -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTWO_RAM_REGIONS "-DDEVICE_PORTIN=1" -DTARGET_RELEASE -DTARGET_STM "-DDEVICE_SERIAL_FC=1" "-DDEVICE_USTICKER=1" "-DDEVICE_TRNG=1" -D__MBED_CMSIS_RTOS_CM "-DDEVICE_SLEEP=1" -DTOOLCHAIN_GCC_ARM "-DDEVICE_SPI=1" "-DDEVICE_SPISLAVE=1" "-DDEVICE_ANALOGIN=1" "-DDEVICE_SERIAL=1" "-DLPTICKER_DELAY_TICKS=3" "-DDEVICE_FLASH=1" -DARM_MATH_CM4 "-DHSE_VALUE=16000000U" "-DMBED_CONF_EVENTS_SHARED_EVENTSIZE              = 256" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE     = 256" "-DMBED_CONF_EVENTS_SHARED_STACKSIZE              = 1024" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE     = 1024" "-DMBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE       = 256   " "-DMBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE       = 256   " "-DMBED_CONF_NSAPI_PRESENT                        = 0     " "-DMBED_CONF_FILESYSTEM_PRESENT                   = 0     " "-DMBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE    = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_BAUD_RATE             = 115200" "-DMBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES      = 0     " "-DMBED_CONF_PPP_CELL_IFACE_APN_LOOKUP            = 0     " "-DMBED_CONF_EVENTS_PRESENT                       = 1     " "-DMBED_CONF_RTOS_PRESENT                         = 1     " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_BUFFER_SIZE = 256   " "-DMBED_CONF_PLATFORM_STDIO_BAUD_RATE             = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_TIMEOUT     = 8000  " "-DMBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT         = 1     " -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/drivers -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/events -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/TARGET_FLASH_CMSIS_ALGO -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/storage_abstraction -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/platform -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Config -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/TARGET_NUCLEO_L486RG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_NUCLEO_L486RG -c C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c -o D:/code/t [... cut ...]
Show binary size.
converting to .bin: tested.elf
converting to .hex: tested.elf
Generate symbol file
Generate list file
Generate OpenOCD config
C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin/arm-none-eabi-size.exe -d -B D:/code/tested/.build/qtc_MeshKit-debug/tested.qtc_MeshKit/tested.elf
   text	   data	    bss	    dec	    hex	filename
  73776	   2668	  10940	  87384	  15558	D:/code/tested/.build/qtc_MeshKit-debug/tested.qtc_MeshKit/tested.elf
14:10:49: Elapsed time: 00:39.```

0xc0170, 2018-06-25:
`mbed compile -m DISCO_L496AG -t GCC_ARM`

(you did not post the command to reproduce, I can see L4 STM32 device there so using on  of these).

```
Compile [ 74.9%]: mbed_rtx_handlers.c
[Warning] mbed_rtx_handlers.c@44,18: unused variable 'tid' [-Wunused-variable]
Compile [ 75.1%]: cmsis_os1.c
Compile [ 75.2%]: irq_cm4f.S
```

No warning from cmsis_os1.c as you can see.  I checked the includes, they look fine. Mbed OS config for RTX header file should get included inside the default values so no redefinition should be present there

leibin2014, 2018-06-26:
@0xc0170 
My target is TARGET_NUCLEO_L486RG. And the warnings are reproduce when compile rtx_evr.c but not cmsis_os1.c. I can use below simple command to reproduce the warnings. I think you can also use it to reproduce(just change the g++ path and change the mbed root from `C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal` to your real path).
```C:/MeshEmbedded-0.0.9/tools/toolchain/bin/arm-none-eabi-g++.exe "-std=gnu++98" "-fno-rtti" "-Wvla" "-c" "-Wall" "-Wextra" "-Wno-unused-parameter" "-Wno-missing-field-initializers" "-fmessage-length=0" "-fno-exceptions" "-fno-builtin" "-ffunction-sections" "-fdata-sections" "-funsigned-char" "-MMD" "-fno-delete-null-pointer-checks" "-fomit-frame-pointer" "-Os" "-g1" "-mcpu=cortex-m4" "-mthumb" "-mfpu=fpv4-sp-d16" "-mfloat-abi=softfp" -D__CORTEX_M4 -DARM_MATH_CM4 "-DMBED_BUILD_TIMESTAMP=1528862553.02" "-DDEVICE_CRC=1" "-DDEVICE_I2C=1" "-D__MBED__=1" "-DDEVICE_I2CSLAVE=1" "-D__FPU_PRESENT=1" "-DDEVICE_PORTOUT=1" -DUSBHOST_OTHER "-DDEVICE_PORTINOUT=1" -DTARGET_RTOS_M4_M7 "-DDEVICE_LOWPOWERTIMER=1" "-DDEVICE_RTC=1" -DTOOLCHAIN_object "-DDEVICE_SERIAL_ASYNCH=1" -D__CMSIS_RTOS "-DDEVICE_ANALOGOUT=1" -DTOOLCHAIN_GCC "-DDEVICE_CAN=1" -DTARGET_STM32L486xG -DTARGET_CORTEX_M "-DDEVICE_I2C_ASYNCH=1" -DTARGET_LIKE_CORTEX_M4 -DTARGET_STM32L486RG -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_STM32L4 "-DDEVICE_LPTICKER=1" "-DDEVICE_SPI_ASYNCH=1" "-DDEVICE_PWMOUT=1" "-DDEVICE_INTERRUPTIN=1" -DTARGET_CORTEX "-DTRANSACTION_QUEUE_SIZE_SPI=2" -D__CORTEX_M4 "-DDEVICE_STDIO_MESSAGES=1" -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTWO_RAM_REGIONS "-DDEVICE_PORTIN=1" -DTARGET_RELEASE -DTARGET_STM "-DDEVICE_SERIAL_FC=1" "-DDEVICE_USTICKER=1" "-DDEVICE_TRNG=1" -D__MBED_CMSIS_RTOS_CM "-DDEVICE_SLEEP=1" -DTOOLCHAIN_GCC_ARM "-DDEVICE_SPI=1" "-DDEVICE_SPISLAVE=1" "-DDEVICE_ANALOGIN=1" "-DDEVICE_SERIAL=1" "-DLPTICKER_DELAY_TICKS=3" "-DDEVICE_FLASH=1" -DARM_MATH_CM4 "-DHSE_VALUE=16000000U" "-DMBED_CONF_EVENTS_SHARED_EVENTSIZE              = 256" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE     = 256" "-DMBED_CONF_EVENTS_SHARED_STACKSIZE              = 1024" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE     = 1024" "-DMBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE       = 256   " "-DMBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE       = 256   " "-DMBED_CONF_NSAPI_PRESENT                        = 0     " "-DMBED_CONF_FILESYSTEM_PRESENT                   = 0     " "-DMBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE    = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_BAUD_RATE             = 115200" "-DMBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES      = 0     " "-DMBED_CONF_PPP_CELL_IFACE_APN_LOOKUP            = 0     " "-DMBED_CONF_EVENTS_PRESENT                       = 1     " "-DMBED_CONF_RTOS_PRESENT                         = 1     " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_BUFFER_SIZE = 256   " "-DMBED_CONF_PLATFORM_STDIO_BAUD_RATE             = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_TIMEOUT     = 8000  " "-DMBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT         = 1     " -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/drivers -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/events -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/TARGET_FLASH_CMSIS_ALGO -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/storage_abstraction -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/platform -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Config -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/TARGET_NUCLEO_L486RG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_NUCLEO_L486RG -c C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c -o D:/code/tested/.build/qtc_MeshKit-debug/mbed-hal.qtc_MeshKit/.obj/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c.o```

leibin2014, 2018-06-26:
The full log:

```C:\Users\bin.lei>C:/MeshEmbedded-0.0.9/tools/toolchain/bin/arm-none-eabi-g++.exe "-std=gnu++98" "-fno-rtti" "-Wvla" "-c" "-Wall" "-Wextra" "-Wno-unused-parameter" "-Wno-missing-field-initializers" "-fmessage-length=0" "-fno-exceptions" "-fno-builtin" "-ffunction-sections" "-fdata-sections" "-funsigned-char" "-MMD" "-fno-delete-null-pointer-checks" "-fomit-frame-pointer" "-Os" "-g1" "-mcpu=cortex-m4" "-mthumb" "-mfpu=fpv4-sp-d16" "-mfloat-abi=softfp" -D__CORTEX_M4 -DARM_MATH_CM4 "-DMBED_BUILD_TIMESTAMP=1528862553.02" "-DDEVICE_CRC=1" "-DDEVICE_I2C=1" "-D__MBED__=1" "-DDEVICE_I2CSLAVE=1" "-D__FPU_PRESENT=1" "-DDEVICE_PORTOUT=1" -DUSBHOST_OTHER "-DDEVICE_PORTINOUT=1" -DTARGET_RTOS_M4_M7 "-DDEVICE_LOWPOWERTIMER=1" "-DDEVICE_RTC=1" -DTOOLCHAIN_object "-DDEVICE_SERIAL_ASYNCH=1" -D__CMSIS_RTOS "-DDEVICE_ANALOGOUT=1" -DTOOLCHAIN_GCC "-DDEVICE_CAN=1" -DTARGET_STM32L486xG -DTARGET_CORTEX_M "-DDEVICE_I2C_ASYNCH=1" -DTARGET_LIKE_CORTEX_M4 -DTARGET_STM32L486RG -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_STM32L4 "-DDEVICE_LPTICKER=1" "-DDEVICE_SPI_ASYNCH=1" "-DDEVICE_PWMOUT=1" "-DDEVICE_INTERRUPTIN=1" -DTARGET_CORTEX "-DTRANSACTION_QUEUE_SIZE_SPI=2" -D__CORTEX_M4 "-DDEVICE_STDIO_MESSAGES=1" -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTWO_RAM_REGIONS "-DDEVICE_PORTIN=1" -DTARGET_RELEASE -DTARGET_STM "-DDEVICE_SERIAL_FC=1" "-DDEVICE_USTICKER=1" "-DDEVICE_TRNG=1" -D__MBED_CMSIS_RTOS_CM "-DDEVICE_SLEEP=1" -DTOOLCHAIN_GCC_ARM "-DDEVICE_SPI=1" "-DDEVICE_SPISLAVE=1" "-DDEVICE_ANALOGIN=1" "-DDEVICE_SERIAL=1" "-DLPTICKER_DELAY_TICKS=3" "-DDEVICE_FLASH=1" -DARM_MATH_CM4 "-DHSE_VALUE=16000000U" "-DMBED_CONF_EVENTS_SHARED_EVENTSIZE              = 256" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_EVENTSIZE     = 256" "-DMBED_CONF_EVENTS_SHARED_STACKSIZE              = 1024" "-DMBED_CONF_EVENTS_SHARED_HIGHPRIO_STACKSIZE     = 1024" "-DMBED_CONF_DRIVERS_UART_SERIAL_RXBUF_SIZE       = 256   " "-DMBED_CONF_DRIVERS_UART_SERIAL_TXBUF_SIZE       = 256   " "-DMBED_CONF_NSAPI_PRESENT                        = 0     " "-DMBED_CONF_FILESYSTEM_PRESENT                   = 0     " "-DMBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE    = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_BAUD_RATE             = 115200" "-DMBED_CONF_PLATFORM_STDIO_CONVERT_NEWLINES      = 0     " "-DMBED_CONF_PPP_CELL_IFACE_APN_LOOKUP            = 0     " "-DMBED_CONF_EVENTS_PRESENT                       = 1     " "-DMBED_CONF_RTOS_PRESENT                         = 1     " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_BUFFER_SIZE = 256   " "-DMBED_CONF_PLATFORM_STDIO_BAUD_RATE             = 9600  " "-DMBED_CONF_PPP_CELL_IFACE_AT_PARSER_TIMEOUT     = 8000  " "-DMBED_CONF_PLATFORM_STDIO_FLUSH_AT_EXIT         = 1     " -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/drivers -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/events -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/TARGET_FLASH_CMSIS_ALGO -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/hal/storage_abstraction -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/platform -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Config -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Include -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/TARGET_CORTEX_M -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4 -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/device -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L486xG/TARGET_NUCLEO_L486RG -IC:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/targets/TARGET_NUCLEO_L486RG -c C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c -o D:/code/tested/.build/qtc_MeshKit-debug/mbed-hal.qtc_MeshKit/.obj/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c.o
In file included from C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/cmsis/RTE_Components.h:22:0,
                 from C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c:31:
C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/mbed_rtx_conf.h:35:0: warning: "OS_STACK_SIZE" redefined
[... cut ...]
 #define OS_TIMER_THREAD_TZ_MOD_ID    1
 ^
In file included from C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Include/rtx_evr.h:30:0,
                 from C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evr.c:28:
C:/Users/bin.lei/git/github/mesh-embedded-rtos/hal/mbed-hal/rtos/TARGET_CORTEX/rtx5/RTX/Config/RTX_Config.h:270:0: note: this is the location of the previous definition
 #define OS_TIMER_THREAD_TZ_MOD_ID   0```

leibin2014, 2018-06-26:
As we can see both `mbed_rtx_conf.h` and `RTX_Config.h(in 'RTE_Components.h')` are include in rtx_evr.c. `OS_STACK_SIZE` is defined in both header file. `RTX_Config.h` is included before `mbed_rtx_conf.h`, and there is no `#ifndef OS_STACK_SIZE` checking in `mbed_rtx_conf.h`, then the redefined warning asserts.

0xc0170, 2018-06-29:
I am fixing rtx_lib include - https://github.com/ARMmbed/mbed-os/pull/7364 . Once I finish that one (changes includes from RTX, might affect this one), I'll look at this one, should not be any warning there

deepikabhavnani, 2018-08-30:
@leibin2014  - Can you confirm if you still get warnings on master branch?

deepikabhavnani, 2018-09-11:
Closing this, feel free to reopen if issue still exists

closed 2018-09-11 by deepikabhavnani