ARMmbed/mbed-os #7306: Online Compiler and ARM GCC build inconsistency?

cederom, 2018-06-23:
Hello world :-)

I am a bit confused as I can see some inconsistency between Online Compiler and offline ARM GCC [armgcc] builds:
* nRF52 BLE UART examples does not build in Online Compiler and the suggestion is to use Offline Compiler [bleuart].
* Offline Compiler does not seem to be able to build even simplest example 
* Below I attach example code with build output errors.

Any hints are welcome :-)

[armgcc] https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
[bleuart] https://os.mbed.com/questions/81587/nRF52_DK-BLE-UART-examples-does-not-buil/
[eventq_ex1] https://os.mbed.com/docs/v5.9/tutorials/the-eventqueue-api.html

[X] Question  
[ ] Enhancement  
[X] Bug  

I have updated the MbedOS to 5.9.1, below is the code example and the build output.. I am building with clean `BUILD` directory just to make sure build is clean..

```
(venv36) uname -a
Darwin iCeDeROM.local 17.6.0 Darwin Kernel Version 17.6.0: Tue May  8 15:22:16 PDT 2018; root:xnu-4570.61.1~1/RELEASE_X86_64 x86_64

(venv36) python --version
Python 3.6.5

(venv36) arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

(venv36) mbed toolchain
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] GCC_ARM

(venv36) mbed target
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] NRF52_DK
```

1. Event queue example:
```
#include "mbed.h"
#include "mbed_events.h"

DigitalOut led1(LED1);
InterruptIn sw(BUTTON2);
EventQueue queue(32 * EVENTS_EVENT_SIZE);
Thread t;

void rise_handler(void) {
    // Toggle LED
    led1 = !led1;
}

void fall_handler(void) {
    printf("fall_handler in context %p\r\n", Thread::gettid());
    // Toggle LED
    led1 = !led1;
}

int main() {
    // Start the event queue
    t.start(callback(&queue, &EventQueue::dispatch_forever));
    printf("Starting in context %p\r\n", Thread::gettid());
    // The 'rise' handler will execute in IRQ context
    sw.rise(rise_handler);
[... 1973 lines cut ...]
[Error] main.cpp@22,2: 'queue' was not declared in this scope
[ERROR] ./source/main.cpp:14:1: error: invalid use of template-name 'rtos::Queue' without an argument list
 Queue queue(8*EVENTS_EVENT_SIZE);
 ^~~~~
./source/main.cpp:14:1: note: class template argument deduction is only available with -std=c++1z or -std=gnu++1z
In file included from ./mbed-os/rtos/Mail.h:28:0,
                 from ./mbed-os/rtos/rtos.h:34,
                 from ./mbed-os/mbed.h:37,
                 from ./source/main.cpp:6:
./mbed-os/rtos/Queue.h:53:7: note: 'template<class T, long unsigned int queue_sz> class rtos::Queue' declared here
 class Queue : private mbed::NonCopyable<Queue<T, queue_sz> > {
       ^~~~~
./source/main.cpp: In function 'int main()':
./source/main.cpp:22:2: error: 'queue' was not declared in this scope
  queue.call_every(1000, blinkCallback);
  ^~~~~
./source/main.cpp:22:2: note: suggested alternative: 'equeue'
  queue.call_every(1000, blinkCallback);
  ^~~~~
  equeue

[mbed] ERROR: "/(...)/lab/venv36/bin/python3.6" returned error code 1.
[mbed] ERROR: Command "/(...)/lab/(...)/lab/venv36/bin/python3.6 -u /(...)/lab/(...)/mbed-os/tools/make.py -t GCC_ARM -m NRF52_DK --source . --build ./BUILD/NRF52_DK/GCC_ARM" in "/(..)/lab/(...)"
---
```

cederom, 2018-06-24:
Update: BLE + UART + nRF52 now builds in Online Compiler, I had to update all internal components/libraries by hand one by one..

Still I am experiencing offline build issues with Queue type and BatteryService..

0xc0170, 2018-06-25:
Use this https://docs.mbed.com/docs/mbedmicro-api/en/latest/ignoring_files_from_build/ to ignore folders/files

0xc0170, 2018-06-25:
And I assume the question above is answered and this issue can be closed , or?

cederom, 2018-06-25:
Thank you @0xc0170, that does the job :-) I thought mbed will only build sources from the configuration locations list, but the logic is opposite here I need to specify which files not to build.. thank you! :-)

closed 2018-06-25 by cederom