ARMmbed/mbed-os #7308: Deep_sleep_lock Ticker.h Issue

tpet93, 2018-06-25:
### Description
In Ticker.h,  **Ticker::attach_us(Callback<void()> func, us_timestamp_t t)** is defined and not just declared.
This causes an issue with the line 119   **sleep_manager_lock_deep_sleep();**

The deep sleep manager tracks sleep locks using the file name.
the deep sleep lock is placed in "Ticker.h" and removed in "Ticker.cpp"
(line 30)
this causes the sleep manager lock counter to lose count,

### Solution
Moving the definition of **Ticker::attach_us(Callback<void()> func, us_timestamp_t t)** 
to  Ticker.cpp from Ticker.h solves the issue.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-06-25:
Thank you for reporting and also providing a solution, we will review soon

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTHAL-233*

mprse, 2019-01-24:
Analysis and patch can be found here: PR https://github.com/ARMmbed/mbed-os/pull/9483.

Fixed as suggested by @tpet93 . 

Thanks @tpet93 !