ARMmbed/mbed-os #7312: mbed should be compilable with "-Werror"

guest253456, 2018-06-25:
I would like to use "-Werror" for my own code but there are too many errors (34617 to be exact) inside mbed. I unfortunately haven't found a way to make GCC ignore errors from files in a given directory.

i would like to use the following flags in my code.

```
"-pedantic","-Wextra", "-Werror=return-type", "-Werror=uninitialized",
 "-Wcast-align", "-Wcast-qual", "-Wctor-dtor-privacy", "-Wdisabled-optimization",
"-Wformat=2", "-Winit-self", "-Wlogical-op", "-Wmissing-declarations", "-Wmissing-include-dirs",
"-Wnoexcept", "-Wold-style-cast", "-Woverloaded-virtual", "-Wredundant-decls", "-Wshadow",
"-Wsign-conversion", "-Wsign-promo", "-Wstrict-null-sentinel", "-Wstrict-overflow=2",
"-Wswitch-default", "-Wundef", "-Wno-unused"
```

mattbrown015, 2018-06-26:
After updating to 5.9.0 I noticed a few warnings appeared in my build using the _normal_ develop profile; I thought this was a bit poor.  I've got quite a lot of files listed in `.mbedignore` so there may well be more than a few warnings.

I was wondering whether to open an issue, open 3 issues or just do nothing.  Up until I saw this issue I was doing nothing and hoping someone else, i.e. the original programmer, would notice and fix them.  

I'm building for STM32L4 and get these warnings:

    Compile [ 51.4%]: mbed_ticker_api.c
    [Warning] mbed_ticker_api.c@47,22: comparison between signed and unsigned integer expressions [-Wsign-compare]
    ...
    Compile [ 54.5%]: mbed_error_hist.c
    [Warning] mbed_error_hist.c@39,5: implicit declaration of function 'memcpy'; did you mean 'fmemopen'? [-Wimplicit-function-declaration]
    ...
    Compile [ 61.7%]: mbed_rtx_handlers.c
    [Warning] mbed_rtx_handlers.c@44,18: unused variable 'tid' [-Wunused-variable]

I've just checked the lastest master, f3424da, and the `mbed_error_hist.c` warning appears to have been fixed.

Having never actually contributed to Mbed OS I did think that fixing a _simple_ warning might be a good way to practise raising issues and completing pull requests but, of course, I haven't found time.  And, this approach isn't going to work for the 34617 warnings found by @guest253456.

While I think library vendors should take pride in ensuring their libraries build without warnings in a large library it can be problematic.  So library users have to trust the library vendor and not worry about the library code.  Perhaps another approach for @guest253456 would be to compile `mbed-os` to a static library and then only the user code is being compiled with `-Werror`.  I've been thinking of trying this approach but not found the time.

Just my 2p, thanks!

ManDeJan, 2018-06-26:
With GCC you can use
[`#pragma GCC system_header`](https://gcc.gnu.org/onlinedocs/cpp/System-Headers.html)
to silence warnings in a header, you could write a header that wraps mbed.h like so:
```
#pragma once

#pragma GCC system_header
#include "mbed.h"
```
I know this is not a solution for mbed, but at least you can use `-Werror` in your own code. 
Hope this helps :)

guest253456, 2018-06-28:
>to silence warnings in a header, you could write a header that wraps mbed.h like so:

This doesn't work for me. Getting new errors.
>mbed-os/rtos/Kernel.h:27: error: unknown type name 'namespace'
> namespace rtos {

0xc0170, 2018-06-28:
Thanks @mattbrown015 for the input. Feel free to contribute, fixing warnings sounds as a good first contributions.

@ARMmbed/mbed-os-test @geky @bulislaw  I am thinking something similar what we have for AStyle where we show how many files actually are not yet conforming to the our coding style (we are slowly fixing them). What if we do the same for warnings.  Any PR that increases number of warnings would be flagged and those should be fixed (reported as +15 warnings (compared to master) thus we wwould know  this is adding 15 more warnings, should be fixed). And we work slowly to clean up warnings.

bulislaw, 2018-06-28:
Shaming people that introduce extra warnings makes sense to me. As for fixing and enabling -Werror globally that may be more problematic as we have some partners code, IP stacks, etc in sources.

geky, 2018-06-28:
>  I am thinking something similar what we have for AStyle where we show how many files actually are not yet conforming to the our coding style

I think this is a great idea :+1:

> As for fixing and enabling -Werror globally that may be more problematic as we have some partners code, IP stacks, etc in sources.

Eventually we want to have every permutation ran in mbed OS's CI no? At that point it'd easy to _find_ the warnings, though maybe intractable to fix. I think it's at leasts worth working towards.

0xc0170, 2018-07-03:
Some ideas that were shared here will be explored by our team (separate from this thread).

> Shaming people that introduce extra warnings makes sense to me. As for fixing and enabling -Werror globally that may be more problematic as we have some partners code, IP stacks, etc in sources.

To answer initial request - we can't currently use `werror` for reasons shared above. We will minimize the warnings on the common codebase (few other tickets are opened related to them). I'll close this as answered. 

@guest253456 feel free to send PR fixing warnings

closed 2018-07-03 by 0xc0170