ARMmbed/mbed-os #7314: Fatal error with mbed-os 5.9.1

DBS06, 2018-06-25:
### Description

toolchain: GNU Tools for ARM Embedded Processors 6-2017-q2-update
mbed-cli: version 1.7.3
mbed-os: 5.9.1

I want to update to mbed-os version 5.9.1 from 5.8.6 and I get this weird error when I build with `mbed compile`:
```
Scan: .
Scan: mbed
Scan: env
Using regions bootloader, application, post_application in this build.
  Region bootloader: size 0x6000, offset 0x0
  Region application: size 0x7b000, offset 0x6000
  Region post_application: size 0x7f000, offset 0x81000
Compile [  0.2%]: mbed_tz_context.c
Compile [  0.4%]: AnalogIn.cpp
Compile [  0.7%]: BusIn.cpp
Compile [  0.9%]: BusInOut.cpp
Compile [  1.1%]: BusOut.cpp
Compile [  1.3%]: Ethernet.cpp
Compile [  1.5%]: CAN.cpp
Compile [  1.7%]: FlashIAP.cpp
Compile [  2.0%]: I2CSlave.cpp
Compile [  2.2%]: I2C.cpp
Compile [  2.4%]: InterruptIn.cpp
Compile [  2.6%]: MbedCRC.cpp
Compile [  2.8%]: InterruptManager.cpp
Compile [  3.0%]: RawSerial.cpp
Compile [  3.3%]: SPISlave.cpp
Compile [  3.5%]: SPI.cpp
Compile [  3.7%]: TableCRC.cpp
Compile [  3.9%]: Serial.cpp
Compile [  4.1%]: Ticker.cpp
Compile [  4.3%]: SerialBase.cpp
Compile [  4.6%]: Timeout.cpp
Compile [  4.8%]: Timer.cpp
Compile [  5.0%]: TimerEvent.cpp
Compile [  5.2%]: equeue.c
Compile [  5.4%]: equeue_posix.c
Compile [  5.7%]: UARTSerial.cpp
Compile [  5.9%]: EventQueue.cpp
Compile [  6.1%]: DeviceKey.cpp
[Fatal Error] c:\Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app\mbed-os\features\device_key\source\DeviceKey.cpp:20:21: nvstore.h: No such file or directory
[ERROR] .\mbed-os\features\device_key\source\DeviceKey.cpp:20:21: fatal error: nvstore.h: No such file or directory
 #include "nvstore.h"
                     ^
compilation terminated.

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u c:\Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app\mbed-os\tools\make.py -t GCC_ARM -m EFM32PG12_STK3402 --profile c:\Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app\profiles\release.json --source . --build BUILD_RELEASE\bootloader-app\EFM32PG12_STK3402 --app-config c:\Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app\ng-apps\bootloader-app\cfg\mbed_app_bootloader.json" in "c:\Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app"
---
```

But `nvstore.h` and `nvstore.cpp` are existing. I don't have any clue why this fails, can anyone help?

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

0xc0170, 2018-06-25:
`Users\phst\Documents\_PROJECTS\C22477_EVVA_EMZY3_VM\_REPOS\ng-master-app` - one thing I would test - shorten paths. move the app closer to root, and retry

DBS06, 2018-06-25:
@0xc0170 Moved the app to `C:\`, but it does not change anything, still got the same error message  😟

DBS06, 2018-06-25:
@0xc0170 Ok found the problem after looking in the git log, someone wrote inside the `.mbedignore`:
```
mbed-os/rtos/*
mbed-os/features/cellular/
mbed-os/features/FEATURE_CLIENT/*
mbed-os/features/FEATURE_COMMON_PAL/*
mbed-os/features/FEATURE_UVISOR/*
mbed-os/features/frameworks/*
mbed-os/features/net/*
mbed-os/features/netsocket/*
mbed-os/features/nvstore/*
mbed-os/features/storage/*
```

closed 2018-06-25 by DBS06