ARMmbed/mbed-os #7315: Requires a method to configure LPUART clock source manually in STM32L4 family.

Eman869, 2018-06-25:
### Description
  Currently in STM32L4 family, LPUART clock source is determined by baud rate. If baud rate is equal or lower than 9600, clock source is set as LSE, otherwise it is PCLK. It would be more flexible that the clock source is selected by user manually. 

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-06-25:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-06-25:
Hi

Thanks for the suggestion. I'll send a PR soon...

bcostm, 2018-06-26:
@Eman869 can you please check if this PR solves your issue ? Thanks.

Eman869, 2018-07-03:
@bcostm Yes, the LPUART clock is able to change manually in compile time.

0xc0170, 2018-07-03:
@Eman869 This should be now resolved. I'll close it

closed 2018-07-03 by 0xc0170