ARMmbed/mbed-os #7317: Requires to configure system clock after back from stop mode manually in STM32L4 family

Eman869, 2018-06-25:
### Description
Currently, MSI is the default system clock after system wake up from stop mode. To wake up system from peripherals like LPUART, I2C, and SWPMI, HSI should be selected. A method to configure HSI as system clock after system wake up from stop mode is required.

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

0xc0170, 2018-06-25:
@ARMmbed/team-st-mcd Please review

bcostm, 2018-06-25:
I think if we test `#if CLOCK_SOURCE == USE_PLL_MSI` instead of testing `TARGET_STM32L4 ||TARGET_STM32L1` it should make the trick.

0xc0170, 2018-07-03:
@Eman869 we need some additional info for this issue

Eman869, 2018-07-10:
@bcostm What needed is not to re-configure clock, but to set the default clock as HSI after exiting STOP mode. Register RCC_CGFR, bit15 is related. MSI is not able to set as clock source for LPUART, I2C, and SWPMI.

screamerbg, 2018-09-10:
@Eman869 @LMESTM Any follow up on this?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5655*

jeromecoutant, 2019-10-11:
@Eman869 Are you expecting some development ?

closed 2019-10-31 by ciarmcom