ARMmbed/mbed-os #7319: I2C byte operations broken on the K64F

c1728p9, 2018-06-25:
### Description

When using byte I2C operations (`i2c_byte_read`, `i2c_byte_write`) on the K64F the device eventually locks up. Attached is an [example main file](https://github.com/ARMmbed/mbed-os/files/2134033/example.zip) which reproduces the problem by communicating wth the I2C accelerometer `MMA7660`. This failure was reproduced on the tip of master which was 8e170ccbd1e22e5545e960061b9dc34976fd0176. This problem likely effect many older versions though.

The callstack at the time of the lockup:
```
i2c_byte_write() at i2c_api.c:224
mbed::I2C::write() at I2C.cpp:79
MMA7660ReadAndPrintXYZ_byte() at main.cpp:51
main() at main.cpp:19
```
I2C traffic of the failure:

[image]

Thanks to @mrcoulter45 for finding this initially.

Note - this problem does not effect the address based I2C functions - `i2c_read` and `i2c_write`

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

c1728p9, 2018-06-25:
CC @ansondtx20 @dreemkiller @mmahadevan108

mmahadevan108, 2018-08-21:
I am unable to reproduce the lockup on the current master. Test prints X, Y, Z readings repeatedly on the serial terminal, I let it run for about 5 minutes.

maclobdell, 2018-09-18:
@c1728p9 - Is this still an issue?

maclobdell, 2018-09-24:
@c1728p9  bump

c1728p9, 2018-09-25:
This is still a problem. @mmahadevan108 I tried b53a9ea4c02fd67cb0cc94d08361e8815585b7bf as you did but it did not work for me. The K64F still hangs when I run the test with the accelerometer connected. Do you get different accelerometer readings when you shake the sensor?

mmahadevan108, 2018-09-27:
I don't see the sensor readings change, always prints X, Y, Z to be 153.

I have confirmed the accelerometer on the board works via a different test app.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5656*

maclobdell, 2019-01-14:
@c1728p9 - do you have hardware that you can demonstrate this issue on?  Can we try to close out this issue by sharing the exact hardware and software that can highlight the problem.  I would be happy to collect the hardware and get it to NXP.

maclobdell, 2019-02-05:
@c1728p9  - how do you recommend we go forward on this one?

c1728p9, 2019-02-05:
@mmahadevan108 in https://github.com/ARMmbed/mbed-os/issues/7319#issuecomment-425201931 you were able to reproduce the failure and also confirm that the accelerometer worked correctly with a different board. Is this sufficient for you to debug and correct this issue?

mrcoulter45, 2019-02-05:
> @mmahadevan108 <https://github.com/mmahadevan108> in #7319 (comment)

Hey Mac,
These files can be used to cause the I2C read failure with the
Grove-3-Axis-Digital-Accelerometer-MMA7660FC on the K64F.

Regards,

Michael Coulter

0xc0170, 2020-10-22:
I'll close this as the drivers were updated since 2018. If the issue persist, the updated report should be provided.

closed 2020-10-22 by 0xc0170