ARMmbed/mbed-os #7321: Retaining a heap margin

RobMeades, 2018-06-25:
### Description

I have an application in which I would like to dynamically allocate memory while ensuring that there is some dynamic memory free for one critical operation.  To do this I have added heap stats to my build with `MBED_HEAP_STATS_ENABLED=1`, then I call `mbed_stats_heap_get()` and check that `reserved_size - current_size` is greater than my desired margin.  However, I find that `malloc()` fails when there is still quite a lot of heap available.  For instance, when `malloc()`ing 1024 byte chunks in a loop, `reserved_size - current_size` is more than 6000 bytes at the point of failure.  What have I misunderstood and is there a better way of achieving my aim?  Target in this case is `UBLOX_C030_U201`.

Here is a test program:

```
#include "mbed.h"

int main()
{
    mbed_stats_heap_t heap;
    int mallocSize = 1024;
    int mallocTotal = 0;
    int x = 0;
    void *pMalloc;

    printf("Starting.\n");

    mbed_stats_heap_get(&heap);

    printf("Start: %d byte(s) heap used (%d reserved), %d byte(s) heap left, malloc()ing in %d byte chunks.\n",
           (int) heap.current_size, (int) heap.reserved_size,
           (int) (heap.reserved_size - heap.current_size), mallocSize);

    while ((pMalloc = malloc(mallocSize)) != NULL) {
        x++;
        mallocTotal += mallocSize;
        mbed_stats_heap_get(&heap);
        printf("%d: %d byte(s) successfully malloc()ed, %d byte(s) heap used, %d bytes heap left.\n",
               x, mallocTotal, (int) heap.current_size,
               (int) (heap.reserved_size - heap.current_size));
    }

    mbed_stats_heap_get(&heap);
    printf("Unable to malloc() %d byte(s), total successfully malloc()ed %d byte(s), %d byte(s) heap used, %d byte(s) heap left.\n",
           mallocSize, mallocTotal, (int) heap.current_size,
           (int) (heap.reserved_size - heap.current_size));
}
```
...and this is its output:

```
Start: 1024 byte(s) heap used (170456 reserved), 169432 byte(s) heap left, malloc()ing in 1024 byte chunks.
1: 1024 byte(s) successfully malloc()ed, 2048 byte(s) heap used, 168408 bytes heap left.
2: 2048 byte(s) successfully malloc()ed, 3072 byte(s) heap used, 167384 bytes heap left.
...
158: 161792 byte(s) successfully malloc()ed, 162816 byte(s) heap used, 7640 bytes heap left.
159: 162816 byte(s) successfully malloc()ed, 163840 byte(s) heap used, 6616 bytes heap left.
Unable to malloc() 1024 byte(s), total successfully malloc()ed 162816 byte(s), 163840 byte(s) heap used, 6616 byte(s) heap left.
```

### Issue request type

[x] Question  
[ ] Enhancement  
[ ] Bug

SeppoTakalo, 2018-06-26:
I have studied this a bit and I'm certain that malloc() stats do not report real values.

Basically the problem is that malloc() internally requires small structures on before, and after, the allocated block for house keeping. This extra is not counted in into those stats, because each compiler has separate malloc() implementation, so it is not portable.

Basically, to see the effect, you could modify your code to do `malloc(1)` which would leave even more data to be shown as "free" before the allocation starts to fail.

On GCC, one way to really track the heap, is to start tracking the top of HEAP as there is `_sbrk()` function, that extends the heap, when malloc does not find free space from existing one. This same does not exist in IAR as the heap size is statically set to the maximum value. I have no information how malloc() works in ARM5 or 6.

RobMeades, 2018-06-26:
Ah, that makes sense.  I suppose that what I could do for my purposes is just try `malloc()`ating an array of the size that is my heap margin.  If it fails I know that I'm at the margin, if it succeeds I free it again.  This would definitely be portable, at least.

closed 2018-06-26 by RobMeades

SeppoTakalo, 2018-06-27:
I reported this behaviour as a bug in https://github.com/ARMmbed/mbed-os/issues/7343