ARMmbed/mbed-os #7334: Mode change of PIN

drahnr, 2018-06-26:
### Description
I have a set of pins which are re=used depending on the operation mode (SPI, I2C, GPIO IRQ, ADC, DigitalOut).

What is the desired way of doing this in mbed os?

If I utilize a PIN as `DigitalOut`, which de-allocates with a scope and afterwards allocate again with `DigitalIn`.
What is the state while it is not allocated? Is it going back to the default state (and where is that defined?)

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-06-27:
> What is the state while it is not allocated? Is it going back to the default state (and where is that defined?)

This is not yet defined, but should be. This will be address by the specs update for all HAL (including those you are using).

@ARMmbed/mbed-os-core Please review

drahnr, 2018-06-27:
It would be greate if there would be a `convert_to<DigitalOut>(PinConvertible)` where all the Pin types derive from the `PinConvertiable` base, and depending ton the Pin this works or doesn't (runtime check) (just a quick idea though)

deepikabhavnani, 2018-09-18:
This is duplicate of https://github.com/ARMmbed/mbed-os/issues/3106.

Since destructor code is empty, deallocation is not done.

closed 2018-09-18 by deepikabhavnani