ARMmbed/mbed-os #7343: Heap stats are not reporting true values

SeppoTakalo, 2018-06-27:
### Description

Heap statistics returned by either [Memory tracing](https://os.mbed.com/docs/v5.9/tutorials/optimizing.html#runtime-memory-tracing) or [Memory statistics](https://os.mbed.com/docs/v5.9/tutorials/optimizing.html#runtime-memory-statistics) report only values reserved by application, without counting into account the overhead added by allocator itself.

This confuses application developers to believe that they are using less memory than what they actually are. It also renders impossible to estimate available free space, or estimate minimal heap space required by application.

For example: `malloc(1)` shows allocation values rising by one byte, but actually 8 bytes have been reserved from the heap.

As a background information, or studying purposes, most common implementation for `malloc()` is the [Doug Lea's Malloc](http://g.oswego.edu/dl/html/malloc.html) which explains the structures used in it internal bookkeeping. This implementation, I believe, is very close to what GCC uses. Notice following:
> ... the smallest allocatable chunk is 16 bytes in systems with 32-bit pointers and 24 bytes in systems with 64-bit pointers.

This is because it has 4 byte boundary markers (size fields), on each end of the allocated block. Also, each block is 8 byte aligned, and sizes are only multiple of 8 for blocks smaller than 512 bytes.

My proposal would be that memory tracing should return the values it is returning now, but this overhead should be documented, and memory statistics should be fixed to return real heap values.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

c1728p9, 2018-08-31:
Hi @SeppoTakalo when I was working on heap stats a while back I noticed that there is overhead as you mentioned. At the time I decided to report the logical value used by an application (not accounting for the allocation overhead). This was for a couple of reasons:

1. Depending on the allocator and settings the overhead may vary for the same application, but the logical value used won't change. Users who want to account for overhead can do so by multiplying overhead per allocation by stats.alloc_cnt.
2. Accounting for the overhead gives the false impression that as long as you make your heap at least as big as the value reported by heap stats then you are guaranteed not to fail allocations. This is not the case, as the heap may need to be much larger to handle fragmentation.

Rather than changing the value reported by the code, what about explaining allocation overhead and fragmentation in a docs?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-207*

deepikabhavnani, 2018-10-05:
Fixed by https://github.com/ARMmbed/mbed-os/pull/7917

closed 2018-10-05 by deepikabhavnani