ARMmbed/mbed-os #7347: Flash memory size reported by "mbed compile" does not take ARMCC's data compression into account.

TeroJaasko, 2018-06-27:
### Description

Flash memory size reported by "mbed compile" does not take ARMCC's data compression into account.

For example compilation of mbed-os-example-blinky for K64F and ARMCC 5, the output is
```
Elf2Bin: mbed-os-example-blinky
+-------------+-------+-------+------+
| Module      | .text | .data | .bss |
+-------------+-------+-------+------+
| [lib]       | 10739 |    16 |  348 |
| anon$$obj.o |    32 |     0 | 1024 |
| main.o      |    60 |     4 |    0 |
| mbed-os     | 22153 |  2545 | 5888 |
| Subtotals   | 32984 |  2565 | 7260 |
+-------------+-------+-------+------+
Total Static RAM memory (data + bss): 9825 bytes
Total Flash memory (text + data): 35549 bytes

Image: ./BUILD/K64F/ARM/mbed-os-example-blinky.bin
```

map-file created by ARMCC tells that the flash size is 33116 bytes:
```
==============================================================================

Image component sizes

      Code (inc. data)   RO Data    RW Data    ZI Data      Debug   
     19600       2416       2704       2560       6912      47412   Object Totals
         0          0         32          0       1024          0   (incl. Generated)
        52          0          7         11          0          0   (incl. Padding)
     10200        446        556         16        348       7652   Library Totals
        12          0          5          0          0          0   (incl. Padding)

==============================================================================

      Code (inc. data)   RO Data    RW Data    ZI Data      Debug   
     29800       2862       3260       2576       7260      14260   Grand Totals
     29800       2862       3260         56       7260      14260   ELF Image Totals (compressed)
     29800       2862       3260         56          0          0   ROM Totals

==============================================================================

    Total RO  Size (Code + RO Data)                33060 (  32.29kB)
    Total RW  Size (RW Data + ZI Data)              9836 (   9.61kB)
    Total ROM Size (Code + RO Data + RW Data)      33116 (  32.34kB)

==============================================================================
```

And actual generated binary is also 33116 bytes:
```
33116 Jun 27 14:36 ./BUILD/K64F/ARM/mbed-os-example-blinky.bin
```

Difference is there mostly as the "RW Data" section is being compressed and mbed compile just adds together the text and data section sizes. 
AFAIK, some minor error is caused by the padding and compiler generated code being left out from mbed compile's stats, but fixing that might require considerably more effort and is perhaps out of scope of this issue.

Environment:
ARMCC: ARM Compiler 5.06 update 6 (build 750)
mbed-cli: 1.7.3
Mbed OS: "03196b2 (HEAD -> mbed-os-5.9, tag: mbed_lib_rev162, tag: mbed-os-5.9.1, origin/mbed-os-5.9) Merge pull request 7228 from ARMmbed/release-candidate"

Note: this is related to https://github.com/ARMmbed/mbed-os/pull/7244

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

theotherjimmy, 2018-06-28:
Thanks for the steps to analyze this failure @TeroJaasko. I don't have time to get to this for this week. @cmonr I would appreciate it if you could pick this one up. If not, I'll find time next week or the week after that.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-215*

SenRamakri, 2019-01-29:
@TeroJaasko  - Is this still an issue which needs to be addressed or is it being addressed as part of RTX .bss changes as mentioned in #7244.

closed 2020-11-20 by ciarmcom

adbridge, 2020-11-20:
This is against the old tools which are frozen. If this is still an issue with Armclang (once the new tools are fully up and running) then a new issue can be raised. Closing this one.