ARMmbed/mbed-os #7358: Handling multiple SPI slaves idiomatically.

drahnr, 2018-06-28:
### Description
In a multi slave setting, it is not clear how the API is supposed to work, since only one SS line can be passed on at a time.
The only solution I see right now, is handling the slave selection process manually.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

drahnr, 2018-07-06:
It would be nice if this could be added to the documentation as an example and explicitly stating that this logic is already part of the `SPI` class rather than to expect people to dig through the SPI implementation details.

SenRamakri, 2018-07-19:
Thanks @drahnr for pointing this out and we will be updating the documentation to capture this. - https://github.com/ARMmbed/mbed-os-5-docs/pull/632
Let is know if you have more questions.

closed 2018-07-19 by SenRamakri