ARMmbed/mbed-os #7360: Asynch serial: read

pilotak, 2018-06-28:
### Description

Lets concider foloowing code 

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

0xc0170, 2018-06-28:
The driver waits for 8 characters. There's no serial event for this (I can't think of what type of event this would be? We do not have time bounded callback here - no timeout).

You can use your own timeout (use Timeout object or other similar functionality). if you haven't received 8 characters within defined time, abort (should flush rx fifo to discard characters that has been received) and restart the read.

pilotak, 2018-06-28:
Ok thank you, maybe timeout could be an enhancement? or possibly fire event when read starts?

kjbracey-arm, 2018-07-05:
I would not recommend the asynchronous serial API as it stands, as it has the limitations you describe. It's ill-suited for general stream comms traffic.

And that API is a HAL API - revising it means revising every platform's HAL. I believe some serial HAL API work is planned, and this is a known issue which would be addressed, but for now I would suggest avoiding the asynchronous API.

Your best bet for serial input is to use `UARTSerial`, which will use the single-character synchronous HAL API from interrupt and present that to you as a stream via the `FileHandle` API. This is used by all network-type drivers like Wifi modules using `AtCmdHandler`, or PPP-based cellular modems, and is solid and pretty efficient, even if not capable of the same sort of theoretical speed as the asynchronous API.

Although I believe the asynchronous HAL API is actually implemented via interrupt-based PIO rather than DMA on most platforms, due to the hardware having similar lack-of-timeout limitations, so the performance difference isn't as large as you might think.

pilotak, 2018-07-05:
Thank you. That explains, why there is no official documentation available. I will have a look on `UARTSerial`, the reason why i used asynch was because it doesn't block code until everything is sent. I send out 255 chars at 9600 which takes ages - i use Modbus protocol

closed 2018-07-05 by pilotak