ARMmbed/mbed-os #7372: Wise1530 inflated binary size

pan-, 2018-06-29:
### Description

Introduction of #7284 has inflated binaries generated of the MTB_ADV_WISE_1530. 

Before (latest): 

```
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |   107 |     0 |   21 |
| [lib]\c.a        | 24993 |  2472 |   89 |
| [lib]\gcc.a      |  3168 |     0 |    0 |
| [lib]\misc       |   252 |    16 |   28 |
| main.o           |    31 |     0 |    0 |
| mbed-os\drivers  |   192 |     4 |  100 |
| mbed-os\features |   108 |     4 |   88 |
| mbed-os\hal      |  1653 |     4 |   68 |
| mbed-os\platform |  2844 |   264 |  221 |
| mbed-os\rtos     | 10438 |   168 | 6073 |
| mbed-os\targets  |  7328 |     4 |  428 |
| Subtotals        | 51114 |  2936 | 7116 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 10052 bytes
Total Flash memory (text + data): 54050 bytes

```

After #7284 : 

```
+-----------------------+--------+-------+-------+
| Module                |  .text | .data |  .bss |
+-----------------------+--------+-------+-------+
| [fill]                |    167 |     0 |    56 |
| [lib]\c.a             |  32659 |  2472 |    89 |
| [lib]\gcc.a           |   3168 |     0 |     0 |
| [lib]\misc            |    252 |    16 |    28 |
| [lib]\nosys.a         |     32 |     0 |     0 |
| [lib]\wiced_drivers.a | 413117 |   112 |  4433 |
| main.o                |     31 |     0 |     0 |
| mbed-os\drivers       |    338 |     4 |   100 |
| mbed-os\events        |   1354 |     0 |     0 |
| mbed-os\features      |    274 |    12 |    48 |
| mbed-os\hal           |   1653 |     4 |    68 |
| mbed-os\platform      |   3519 |   260 |   217 |
| mbed-os\rtos          |  13445 |   168 |  6073 |
| mbed-os\targets       |   7328 |     4 |   428 |
| Subtotals             | 477337 |  3052 | 11540 |
+-----------------------+--------+-------+-------+
Total Static RAM memory (data + bss): 14592 bytes
Total Flash memory (text + data): 480389 bytes
```

The code to get those results was compiled with GCC: 

```c++
#include "mbed.h"

int main() {
    printf("Hello World!\r\n");
    return 0;
}
```

The cause seems to be a global Non POD object in the library added: 

```sh
arm-none-eabi-objdump.exe -t -r --section .init_array libwiced_drivers.a

wiced_emac.o:     file format elf32-littlearm

SYMBOL TABLE:
00000000 l    d  .init_array    00000000 .init_array

RELOCATION RECORDS FOR [.init_array]:
OFFSET   TYPE              VALUE
00000000 R_ARM_TARGET1     _GLOBAL__sub_I__ZN10WICED_EMACC2Ev
```

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

pan-, 2018-06-29:
@KariHaapalehto Could you look after this issue ?

0xc0170, 2018-07-03:
@mikaleppanen is looking at this issue

mikaleppanen, 2018-07-04:
Fixed in: https://github.com/ARMmbed/mbed-os/pull/7413

closed 2018-07-05 by cmonr