ARMmbed/mbed-os #7379: Heap issue on DISCO_L475VG_IOT01A with simple threading example

Brools, 2018-06-30:
### Description

Hardware: DISCO_L475VG_IOT01A
Compiler: GNU Tools ARM Embedded\6 2017-q2-update
mbed-cli: 1.5.0

I'm having issues with the first and very basic thread example in the mbed reference:
https://os.mbed.com/docs/v5.9/reference/thread.html

For some reason, it appears that the heap allocation for the thread (during thread.start) is failing and causes mbed OS error fault.
The serial output is shown below

**DISCO_L475VG_IOT01A, GCC_ARM (GNU Tools ARM Embedded\6 2017-q2-update\bin)**:
++ MbedOS Error Info ++
Error Status: 0x8001011f Code: 287 Module: 1
Error Message: Operator new[] out of memory

Location: 0x8000f0b
Error Value: 0x1000
Current Thread: Id: 0x2000168c Entry: 0x8001911 StackSize: 0x1000 StackMem: 0x200016d8 SP: 0x20002650 
-- MbedOS Error Info -

I have tried to add the mbed-memory-status from janjongboom, but it did not help me find a solution.
The serial output is shown below

**DISCO_L475VG_IOT01A, GCC_ARM (GNU Tools ARM Embedded\6 2017-q2-update\bin)**:
   stack ( start: 200016D8 end: 200026D8 size: 00001000 used: 000000A4 ) thread ( id: 2000168C entry: 08001C29 name: main_thread )
    stack ( start: 20001318 end: 20001518 size: 00000200 used: 00000040 ) thread ( id: 20000FCC entry: 08001DC1 name: unknown )
    stack ( start: 20001018 end: 20001318 size: 00000300 used: 00000068 ) thread ( id: 20000F84 entry: 08004169 name: unknown )
     heap ( start: 20002A38 end: 20017C00 size: 000151C8 used: 00000000 )  alloc ( ok: 00000000  fail: 00000000 )
isr_stack ( start: 20017C00 end: 20018000 size: 00000400 used: 00000400 )

++ MbedOS Error Info ++
Error Status: 0x8001011f Code: 287 Module: 1
Error Message: Operator new[] out of memory

Location: 0x8001223
Error Value: 0x1000
Current Thread: Id: 0x2000168c Entry: 0x8001c29 StackSize: 0x1000 StackMem: 0x200016d8 SP: 0x20002650 
-- MbedOS Error Info -

Trying with the arm ds-5 compiler 5, it works just fine (excluding the mbed-memory-status)
**DISCO_L475VG_IOT01A, ARM (DS-5_v5.28.1\sw\ARMCompiler5.06u6)**:
works fine

I can add that it works on nrf52_dk with arm compiler 5 AND gcc.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

adustm, 2018-07-02:
Hello Arm team, 
I 've notived the same kind of problem. DISCO_L475VG_IOT01A is crashing since release 5.9.0 (mbed-os-wifi example for instance, many basic mbed-os examples are also crashing)

release 5.8.6 works fine

I think it might be related to the splitted heap and stack memories (1 on SRAM1 and 1 on SRAM2).

adustm, 2018-07-02:
@0xc0170 I did not follow closely the modifications done on the 2 last releases. I have no idea where it comes from.
I've been able to see the results of release 5.9.0 : there are regressions for GCC_ARM toolchain only.

0xc0170, 2018-07-02:
Thanks, we will review

jeromecoutant, 2018-07-02:
Hi

TWO_RAM_REGIONS macro has disappeared since #7009 ...

0xc0170, 2018-07-02:
@jeromecoutant Please send PR. I can confirm, comparing 5.8 to 5.9 - the macro is not present . 
Assuming bad rebase in 7009

adustm, 2018-07-10:
Hello,
Problem fixed on my side, thanks @jeromecoutant 
@Brools is it also ok for you ? Ok to close the issue ?
Kind regards

closed 2018-07-10 by 0xc0170

Brools, 2018-07-16:
Tried the same example with updated master branch. Issue is no longer there - good job :)