ARMmbed/mbed-os #7380: set_time not working on LPC1768

janfwesterkamp, 2018-06-30:
### Description
Using the time example from https://os.mbed.com/docs/latest/reference/time.html and updating mbed-os 
to the newest version (03196b2) in the mbed online compiler returns the following in the terminal:

> Time as seconds since January 1, 1970 = -1
> Time as a basic string = Sun Jan  1 00:00:00 1900
> Time as a custom formatted string = 12:00 AM

The expected result would have been:
> Time as seconds since January 1, 1970 = 1256729737
> Time as a basic string = Wed Oct 28 11:35:37 2009
> Time as a custom formatted string = 11:35 AM

Also not working is revision 866850a (5.9.0).
The last working revision with the expected result is 35fa909 (5.8)

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-07-02:
@ARMmbed/team-nxp Please review

Jossie90, 2018-08-28:
I stumbled upon this issue when updating to the latest mbed release (5.9.0.). I think it is caused by this commit 6d90c2716d64343bb34376a4a9effacc7a7fca99, where the RTC is was disabled for this device. Adding "RTC" to targets.json solved it for me.

0xc0170, 2018-08-28:
RTC should be reenabled (but needs to make RTC testing green first)

mmahadevan108, 2018-09-05:
Below PR to reenable RTC on LPC1768
https://github.com/ARMmbed/mbed-os/pull/8009

kegilbert, 2018-09-05:
If RTC is disabled it'll default to using a ticker which has a known issue and pending fix (haven't looked closely into this yet, just saw the issue and it sounds like it may be related):
https://github.com/ARMmbed/mbed-os/pull/7849

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5784*

janfwesterkamp, 2018-10-10:
With revision c53d51f it's working now as expected - for me.

closed 2018-10-10 by janfwesterkamp