ARMmbed/mbed-os #7381: wait taking 10x longer on LPC 1768

janfwesterkamp, 2018-06-30:
### Description
The wait(), wait_ms(), wait_us(), and Thread::wait() functions take 10 times as long as they should on a LPC 1768 with the online compiler. 
A wait_ms(100) should wait for 100 milliseconds but acutally takes 1 second.

Tested with values for 1 second and 1/10 second.

Tested revisions: 
- 03196b2 (5.9.1)
- 35fa909 (5.8.6)
- c9e63f1 (5.5.1)

### Test program
I used the same program with OS 2 and OS 5 with Thread.wait() only on OS 5.

``` cpp
#include "mbed.h"
DigitalOut myled(LED1);
int main() {
    printf("clock = %d\r\n",SystemCoreClock);    
	printf("mbed-os-rev: %d.%d.%d   lib-rev: %d\r\n", MBED_MAJOR_VERSION, MBED_MINOR_VERSION,   MBED_PATCH_VERSION, MBED_LIBRARY_VERSION);
     
    while(1) {
        myled = 1;
        wait_ms(100);
        //Thread::wait(100);
        myled = 0;
        wait_ms(100);
        //Thread::wait(100);
   }
}
```

#### Terminal output of failed test with OS 5
> clock = 96000000
> mbed-os-rev: 5.9.1   lib-rev: 162

#### Terminal output of successful test with OS 2
> clock = 96000000
> mbed-os-rev: 2.0.159   lib-rev: 159

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-07-02:
@ARMmbed/team-nxp Please review

janfwesterkamp, 2018-07-03:
When I saw #7397, I repeated the test and measured the time with an oscilloscope. Nothing has changed, wait_ms(100) takes exactly 1 second with OS 5 and 0.1 second with OS 2

fasthazard, 2018-08-24:
Just posted a possible solution at #6407, related to older interface firmware revision, although I was only able to test with mbed-cli.

In that case, updating the [lpc1768 interface firmware](https://os.mbed.com/handbook/Firmware-LPC1768-LPC11U24) resolved the error in wait time without needing to reflash my code.

0xc0170, 2018-08-24:
@janfwesterkamp Please test

janfwesterkamp, 2018-08-25:
Updating the firmware fixes the problem. Tested it with 3 different LPC1768, the online compiler and the latest revisions of OS 2 (169:a7c7b63) and OS 5 (0fdfcf735).

There should be firmware related documentation in the new OS 5 handbook.

closed 2018-08-25 by janfwesterkamp