ARMmbed/mbed-os #7392: Bootloader header size causing VTOR to be misaligned

jorisa, 2018-07-02:
### Description
The lowest 6 bits of The Vector Table Offset Register are reserved:
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/Ciheijba.html

When building an application with a header that is smaller than 0x80 this will cause `NVIC_FLASH_VECTOR_ADDRESS` to have an offset that will be truncated assigning this to the VTOR register, causing the header itself to be copied into the vector table.

When following the configuration described in this tutorial the resulting header size is 0x40:
https://os.mbed.com/docs/v5.9/tutorials/bootloader.html
```
Using regions bootloader, header, application in this build.
  Region bootloader: size 0x40000, offset 0x8000000
  Region header: size 0x3c, offset 0x8040000
  Region application: size 0x1bffc0, offset 0x8040040
```

This was tested with mbed-os-5.9.1 (03196b2) on TARGET_NUCLEO_F767ZI, using gcc-arm-none-eabi-6-2017-q2-update-win32.

A workaround is by adding some junk fields in the `target.header_format` in `mbed_lib.json`. A possible fix is to pad the header to multiples of 0x80 in the build script.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

theotherjimmy, 2018-07-03:
@jorisa The vector table is copied to RAM on boot of Mbed OS, and the VTOR is updated to point to that. This is not a problem.

jorisa, 2018-07-04:
@theotherjimmy yes, but...
```c
    uint32_t *old_vectors = (uint32_t *)SCB->VTOR;
    uint32_t *vectors = (uint32_t*)NVIC_RAM_VECTOR_ADDRESS;
    for (int i = 0; i < NVIC_NUM_VECTORS; i++) {
        vectors[i] = old_vectors[i];
    }
    SCB->VTOR = (uint32_t)NVIC_RAM_VECTOR_ADDRESS;
```

`mbed_cpy_nvic` ued the VTOR register as the source address of this copy operation, so the problem remains :)

theotherjimmy, 2018-07-05:
@jorisa Thanks.

Could you test this patch?
```diff
diff --git a/tools/config/__init__.py b/tools/config/__init__.py
index 7162e88ac..20ce90fb2 100644
--- a/tools/config/__init__.py
+++ b/tools/config/__init__.py
@@ -668,7 +668,7 @@ class Config(object):
         size = self._header_size(header_format)
         region = Region("header", start, size, False, None)
         start += size
-        start = ((start + 7) // 8) * 8
+        start = ((start + (2**5 - 1)) // (2**5)) * (2**5)
         return (start, region)
 
     @staticmethod
```

jorisa, 2018-07-06:
I think it needs to be: `start = ((start + (2**7 - 1)) // (2**7)) * (2**7)`

theotherjimmy, 2018-07-06:
Ah, yes. My mistake.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-220*

PR #8270 (2018-10-11): Tools: Ensure vtor reserved bits are all 0
### Description

The VTOR reserves the lowest 7 bits. This PR changes the round up
behavoir of the application offset to make sure that the address used 
for the in-flash vector table always ends in 7 0's.

Fixes #7392

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
commit 7a97f480 (2018-10-11): Tools: Ensure vtor reserved bits are all 0

### Description

The VTOR reserves the lowest 7 bits. This PR changes the round up
behavoir of the application offset to make sure that the address used
for the in-flash vector table always ends in 7 0's.

Fixes #7392

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Breaking change
closed 2018-10-11 by cmonr