ARMmbed/mbed-os #7394: Build tools should generate an application+header binary

jorisa, 2018-07-02:
### Description
When building a application+header to be used with a bootloader, the build script generates a binary file that combines all three images:

However when deploying an application to be used with the bootloader you typically want a binary file that contains the application with header, without the bootloader.

A workaround is mangling the files manually afterwards, but it would be nicer if this was handled natively by the build script. I currently have hacked things up by calling the merge_region_list twice, but this generates an excess `<appname>_withHeader_header.hex`

```python
        # Link Program
        if toolchain.config.has_regions:
            res, _ = toolchain.link_program(resources, build_path, name + "_application")
            region_list = list(toolchain.config.regions)
            region_list = [r._replace(filename=res) if r.active else r
                           for r in region_list]
            res = "%s_withHeader.%s" % (join(build_path, name),
                             getattr(toolchain.target, "OUTPUT_EXT", "bin"))
            merge_region_list(region_list[1:], res, notify)
            res = "%s.%s" % (join(build_path, name),
                             getattr(toolchain.target, "OUTPUT_EXT", "bin"))            
            merge_region_list(region_list, res, notify)
```

### Issue request type
[ ] Question  
[X] Enhancement  
[ ] Bug

theotherjimmy, 2018-07-03:
@jorisa Why do you need this?

theotherjimmy, 2018-07-03:
>  you typically want

Could you be more concrete?

jorisa, 2018-07-04:
@theotherjimmy : When you have a product with a bootloader in the factory you flash it with an image that contains bootloader+(app+header), when you deploy a firmware update you need (app+header) since the header contains information about the image that is consumed by the bootloader.

theotherjimmy, 2018-07-09:
@jorisa Do you need the (app+header) exactly as it would be created in the "include everything" build?

jorisa, 2018-07-20:
Yeah that would work, additionally see #7563

theotherjimmy, 2018-07-20:
I'll make an "update" image available.

closed 2018-07-26 by cmonr

Reda-RM, 2020-01-01:
@theotherjimmy  What was the purpose of removing the header again from the *_update binary at commit #545553b6 ??
How to generate a binary which includes the header to send to the device while doing FOTA?

theotherjimmy, 2020-01-02:
@Reda-RM I recommend a few things:
 * Take a look at the PR that introduced those changes.
 * Open a fresh issue, as others will likely see it.

Further, note that I'm no longer working on Mbed OS, and can't really provide help for these issues mostly because I don't remember the context around this decision.

Reda-RM, 2020-01-05:
Thanks for your reply :)

PR #7567 (2018-07-26): Tools: Generate update images with managed bl mode
### Description

This PR add generation of update images, which do not include the 
bootloader, to managed bootloader mode builds. The output will include
`Update Image: <image_name>` to tell you where the update image is 
located.

Resolves #7394 

### Pull request type

    [ ] Fix
    [ ] Refactor
    [ ] New target
    [x] Feature
    [ ] Breaking change