ARMmbed/mbed-os #7398: UART not working on LPC54608

dreemkiller, 2018-07-02:
### Description

The UART interface on the LPC54608 board does not **appear** to work.
I am attempting to use the WizNet WizFi310 shield (https://os.mbed.com/components/WizFi310-EVB/), and have successfully used it on a K64F.

My code is as follows:
```
void init() {
  WizFi310Interface net(MBED_CONF_APP_WIFI_TX, MBED_CONF_APP_WIFI_RX, true);
  net.connect(MBED_CONF_APP_WIFI_SSID,
                     MBED_CONF_APP_WIFI_PASSWORD,
                     NSAPI_SECURITY_WPA_WPA2, 0);
   const char *ip = net.get_ip_address();
}
```
MBED_CONF_APP_WIFI_TX is "D1"
MBED_CONF_APP_WIFI_RX is "D0"
MBED_CONF_APP_WIFI_* are set to the appropriate values for my network.

The call to net.connect times out on all communications with the shield.
When I connect a logic analyzer, there is no activity on the D0 or D1 mins.
The call to net.get_ip_address() returns NULL (as would be expected if connect failed)

I am compiling with arm-none-eabi-gcc, version 6.3.1 20170620
I am using the mbed cli.
git log -n1 --oneline:
```
ae6c7c60f (HEAD -> mbed-os-5.8, tag: mbed-os-5.8.4, tag: latest, origin/mbed-os-5.8) Merge pull request #6818 from ARMmbed/release-candidate
```

### Issue request type
[ ] Question  
[ ] Enhancement  
[X ] Bug

0xc0170, 2018-07-02:
@ARMmbed/team-nxp Please review

mmahadevan108, 2018-07-02:
Can you try this fix.
[0001-LPC546XX-Fix-UART-MUX-setting.patch.txt](https://github.com/ARMmbed/mbed-os/files/2156565/0001-LPC546XX-Fix-UART-MUX-setting.patch.txt)

mmahadevan108, 2018-07-02:
Please refer to PR https://github.com/ARMmbed/mbed-os/pull/7403

dreemkiller, 2018-07-05:
Is there any reason this change would disable/affect DAPLink?  When I run with this patch, I can’t see anything on the console (that was working before). It’s hard to see the effect of the fix without it, since to my knowledge, PyOCD doesn’t work on this board for 5.8).

Thanks,
D

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

dreemkiller, 2018-07-05:
Also it appears to be affecting the LCD - nothing's showing on it when I use the the patch (and it does without the patch). The patch is the only difference between the two runs.

dreemkiller, 2018-07-16:
Is there any update on this? Do you guys need more information?

dreemkiller, 2018-09-13:
ping

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5782*

dreemkiller, 2018-12-03:
I believe that I've found the problem.
In the serial_api.c file, when:
```C
static inline void uart_irq(uint32_t transmit_empty, uint32_t receive_full, uint32_t index)
{
    if (serial_irq_ids[index] != 0) {
        if (transmit_empty)
            irq_handler(serial_irq_ids[index], TxIrq);

        if (receive_full)
            irq_handler(serial_irq_ids[index], RxIrq);
    }
}
```
```C
void uart4_irq()
{
    uint32_t status_flags = USART0->FIFOSTAT;
    uart_irq((status_flags & kUSART_TxFifoEmptyFlag), (status_flags & kUSART_RxFifoFullFlag), 0);
}
```
If the hardware receives a short UART message, the Rx FIFO will never be full, and the interrupt with fire but never be serviced.
Changing the code to as follows solves my problems (it may not be a complete solution, I leave it to the hardware experts to judge):
```C
static inline void uart_irq(uint32_t transmit_empty, uint32_t receive_not_empty, uint32_t index)
{
    if (serial_irq_ids[index] != 0) {
        if (transmit_empty)
            irq_handler(serial_irq_ids[index], TxIrq);

        if (receive_not_empty)
            irq_handler(serial_irq_ids[index], RxIrq);
    }
}
```
```C
void uart0_irq()
{
    uint32_t status_flags = USART0->FIFOSTAT;
    uart_irq((status_flags & kUSART_TxFifoEmptyFlag), (status_flags & kUSART_RxFifoNotEmptyFlag), 0);
}
```
I will be submitting a PR with these changes for people to peruse.

maclobdell, 2019-01-14:
the PR is merged.  Can this issue be closed?

dreemkiller, 2019-01-14:
I opened it, and I'm satisfied. I can't speak for others, though.

closed 2019-01-15 by 0xc0170