ARMmbed/mbed-os #7400: What could cause timer thread stack overflow?

LiyouZhou, 2018-07-02:
[ ] Question  

I am debugging  MSIS-RTOS error: Stack overflow (status: 0x1, task ID: 0x20005C04, task name: (null))
which comes from the osRtxTimerThread. What are the possible things that could have caused timer thread stack usage increase?
1. Too many threads?
2. Too many Timers?
3. Too many Tickers?

Any more suggestions on what I can look at?

0xc0170, 2018-07-03:
Timer thread size is configured via `MBED_CONF_APP_TIMER_THREAD_STACK_SIZE`. By default its 768.

It could be many timers,  as its indicated by the handbook (see this doc that added info about the config https://github.com/ARMmbed/mbed-os-5-docs/pull/510)

bulislaw, 2018-07-03:
Hm I would be interesting to see how many timers would cause that, 1k of stack is a lot...

SenRamakri, 2018-07-17:
Hi @LiyouZhou, For each timer created using osTimerNew(or some wrapper over it), we can also add a callback function and this callback gets called in osRtxTimerThread. And if such callback functions are using too much of stack you may see the stack overflow on timer thread. In order to debug your issue further I would suggest you to put a breakpoint in https://github.com/ARMmbed/mbed-os/blob/master/rtos/TARGET_CORTEX/mbed_rtx_handlers.c#L50 and analyze the callstack/backtrace. Hope this helps with debugging your situation. Hope this answers your questions and at this point I would like to close this issue. Let us know if you need more help debugging the situation.

closed 2018-07-17 by SenRamakri