ARMmbed/mbed-os #7407: Are FILESYSTEM tests available with small targets ?

jeromecoutant, 2018-07-03:
### Description

Hi

I think some RAM size check should be included in features/tests/filesystem tests ?

Ex:
mbed test -t GCC_ARM -m NUCLEO_F070RB -v -n features-tests-filesystem-*

+-----------------------+---------------+----------------------------------------------------+---------+--------------------+-------------+
| target                | platform_name | test suite                                         | result  | elapsed_time (sec) | copy_method |
+-----------------------+---------------+----------------------------------------------------+---------+--------------------+-------------+
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-buffered_block_device    | TIMEOUT | 46.31              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-fat_filesystem           | TIMEOUT | 27.09              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-flashsim_block_device    | FAIL    | 15.67              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-heap_block_device        | FAIL    | 16.43              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-mbr_block_device         | FAIL    | 15.92              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-multipart_fat_filesystem | TIMEOUT | 27.65              | default     |
| NUCLEO_F070RB-GCC_ARM | NUCLEO_F070RB | features-tests-filesystem-util_block_device        | TIMEOUT | 25.93              | default     |
+-----------------------+---------------+----------------------------------------------------+---------+--------------------+-------------+

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

jeromecoutant, 2018-09-11:
@davidsaada

davidsaada, 2018-09-11:
@jeromecoutant This problem has been fixed in [#7465](https://github.com/ARMmbed/mbed-os/pull/7465).
As it has only been merged on August 1 (while this issue had been opened earlier), I suggest you retest it with master (or any release starting 5.9.5).

jeromecoutant, 2018-09-11:
I am currently checking this result:

| target                  | platform_name       | test suite                                                 | result | elapsed_time (sec) | copy_method |
|-------------------------|---------------------|------------------------------------------------------------|--------|--------------------|-------------|
| DISCO_L072CZ_LRWAN1-IAR | DISCO_L072CZ_LRWAN1 | features-storage-tests-filesystem-fat_filesystem           | FAIL   | 19.9               | default     |
| DISCO_L072CZ_LRWAN1-IAR | DISCO_L072CZ_LRWAN1 | features-storage-tests-filesystem-multipart_fat_filesystem | FAIL   | 19.79              | default     |

Thx

davidsaada, 2018-09-11:
Right. Indeed [#7465](https://github.com/ARMmbed/mbed-os/pull/7465) only dealt with BD tests, not with file system ones. Will need to open an additional PR for that.

jeromecoutant, 2018-09-17:
> Will need to open an additional PR for that.

any update ?

davidsaada, 2018-09-17:
> any update ?

Sorry for not updating. Went over the FAT tests, and they seem to be quite hard to modify for this purpose. The current FAT tests require quite block devices on the one hand, but the format process only allocates small chunks of the memory on demand (which is really very FAT specific). Now, if we try to allocate the full size of the Heap Block Device in advance (to see if we have enough memory), the test will fail for all platforms (even the ones that can run the tests OK). Otherwise, we face the problem with the smaller devices.
@geky What do you say?

geky, 2018-09-19:
Hmm, could we find out how much RAM is used by the HeapBlockDevice after running the FAT tests and then use that as our threshhold (with some margin)?

jeromecoutant, 2018-09-24:
So, there will be no update for file system tests ?

Should I push a PR with like this :

#if defined(TARGET_xxx)
    #error [NOT_SUPPORTED] Insufficient RAM size
#endif

davidsaada, 2018-09-25:
@jeromecoutant Sorry for not being responsive. On holiday vacation until 2.10.
@geky's solution seems like the most appropriate one, even if not optimal. Will implement it when I come back from the holidays. As for your suggestion, we prefer not to add target specific code to our tests right now. We'll probably revise our testing system to have a few levels of testing based on board types, but this is more for the long term.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTSTOR-476*

jeromecoutant, 2018-12-04:
Hi
Any update ?
Thx

davidsaada, 2018-12-06:
@jeromecoutant I'm on it now. Didn't have too much time to handle this for 5.11, due to more prioritized issues.

closed 2018-12-17 by cmonr