ARMmbed/mbed-os #7408: Separate CAN logical layer from the existance of CAN device

OmerRosler, 2018-07-03:
We have a situation where we use a CAN MAC that sends CAN messages over SPI to the microcontroller.

We want to use the mbed CAN api (`CANMessage` is the main type) for this as we use it elsewhere in our codebase. 
The problem is the logical layer of CAN is  protected by `#if defined(DEVICE_CAN)`

Would it be possible to separate this into a separate, unprotected header?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-219*

SenRamakri, 2018-12-04:
Hello @OmerRosler - We looked into this requirement and do think its a great idea. But unfortunately we don't have any immediate plans to separate CAN logical layer. But feel free to provide an implementation for logical layer separation and we would be happy to review and integrate. At present we are closing this issue.

closed 2018-12-04 by SenRamakri