ARMmbed/mbed-os #7409: cryptic compiler errors with GCC_ARM 6

bmcdonnell-ionx, 2018-07-03:
### Description

Cryptic GCC_ARM compiler errors (build failures) when defining a class member function called `putchar()`. (Unknown what other circumstances may produce the same error.) Demo project attached below.

Switching to the latest compiler version (GNU Arm Embedded Toolchain Version 7-2018-q2-update) resolves the issue for me. (That is, I installed it, updated `GCC_ARM_PATH` in `mbed_settings.py` to point to its `bin` directory, ran `python mbed_settings.py`, then `mbed compile`.)

I think this is a compiler issue, but it's not clear to me what the command line is being passed to `arm-none-eabi-c++`, so I'm not sure how to drill down further.

I would think the solution is to update `mbed-os` build tools and [version check](https://github.com/ARMmbed/mbed-os/pull/7247) to use/require GNU Arm Embedded Toolchain version 7. (NB: version check currently states that only version 6 is supported, but that's where I had the problem.)

- Build tools: Mbed CLI (`mbed compile`)
- Targets: ALL
- Toolchain: GCC_ARM 6 (GNU Arm Embedded Toolchain: 6-2017-q2-update June 28, 2017)
- Mbed OS SHA: `62f8b922b (HEAD, tag: mbed-os-5.9.2, origin/mbed-os-5.9) Merge pull request #7375 from ARMmbed/release-candidate`

#### Steps to reproduce
- If you haven't yet, install GCC_ARM version 6 ([GNU Arm Embedded Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads): 6-2017-q2-update June 28, 2017)
- Download and extract this attached example: [mbed-os-emb-cli-cryptic-build-err.zip](https://github.com/ARMmbed/mbed-os/files/2159997/mbed-os-emb-cli-cryptic-build-err.zip)
```
cd mbed-os-emb-cli-cryptic-build-err/
mbed deploy
mbed compile -m <pick a target>
```

#### Result

```
me@pc MINGW64 /c/dev_temp/mbed-os-emb-cli-cryptic-build-err (master)
$ mbed compile -m LPC4088
Building project mbed-os-emb-cli-cryptic-build-err (LPC4088, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Compile [100.0%]: main.cpp
[Error] main.cpp@8,9: '_impure_ptr' is not a type
[Error] main.cpp@8,9: expected identifier before '(' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: '_stdout' does not name a type
[Error] main.cpp@8,9: trailing return type only available with -std=c++11 or -std=gnu++11
[Error] main.cpp@8,9: expected ';' at end of member declaration
[Error] main.cpp@8,9: expected unqualified-id before ')' token
[ERROR] In file included from c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\arm-none-eabi\include\stdio.h:60:0,
                 from .\main.cpp:1:
.\main.cpp:8:9: error: '_impure_ptr' is not a type
    void putchar(char c);
         ^
In file included from .\main.cpp:1:0:
.\main.cpp:8:9: error: expected identifier before '(' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
.\main.cpp:8:9: error: '_stdout' does not name a type
    void putchar(char c);
         ^
.\main.cpp:8:9: error: trailing return type only available with -std=c++11 or -std=gnu++11
.\main.cpp:8:9: error: expected ';' at end of member declaration
.\main.cpp:8:9: error: expected unqualified-id before ')' token
    void putchar(char c);
         ^

[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM" in "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err"
---
```

### Issue request type

[ ] Question
[ ] Enhancement
[X] Bug

0xc0170, 2018-07-04:
What is shown with v7 ?

> I think this is a compiler issue, but it's not clear to me what the command line is being passed to arm-none-eabi-c++, so I'm not sure how to drill down further.

You should get all commands with verbose mode.

deepikabhavnani, 2018-07-05:
@bmcdonnell-ionx 
Issue with Standard library functions (printf / scanf) was found with GNU ARM toolchain  6-2017-q2-update  and 7-2017-q4-major. Issue was root caused to use of C99 IO Formatting with newlib.

GNU Arm Embedded Toolchain Version 7-2018-q2-update release notes mention about the same
"Enabled C99 IO format specifier in newlib"

bmcdonnell-ionx, 2018-07-05:
@deepikabhavnani,

> Issue was root caused to use of C99 IO Formatting with newlib.

How is that related to the issue I've shown here?

And what do you suggest for a corrective course of action?

bmcdonnell-ionx, 2018-07-05:
@0xc0170,

```
me@pc MINGW64 /c/dev_temp/mbed-os-emb-cli-cryptic-build-err (master)
$ mbed compile -vv -m LPC4088
Building project mbed-os-emb-cli-cryptic-build-err (LPC4088, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Macros: -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1530825130.26 -DARM_MATH_CM4
Compile [100.0%]: main.cpp
Compile: arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -D_RTE_ -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DMBED_BUILD_TIMESTAMP=1530825130.26 -DARM_MATH_CM4 @.\BUILD\LPC4088\GCC_ARM\.includes_eec51d9479a32bdfc5a7b9e56c0c7248.txt -include .\BUILD\LPC4088\GCC_ARM\mbed_config.h -MD -MF BUILD\LPC4088\GCC_ARM\main.d -o BUILD\LPC4088\GCC_ARM\main.o .\main.cpp
[Error] main.cpp@8,9: '_impure_ptr' is not a type
[Error] main.cpp@8,9: expected identifier before '(' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: '_stdout' does not name a type
[Error] main.cpp@8,9: trailing return type only available with -std=c++11 or -std=gnu++11
[Error] main.cpp@8,9: expected ';' at end of member declaration
[Error] main.cpp@8,9: expected unqualified-id before ')' token
[DEBUG] Return: 1
[DEBUG] Output: In file included from c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\arm-none-eabi\include\stdio.h:60:0,
[DEBUG] Output:                  from .\main.cpp:1:
[DEBUG] Output: .\main.cpp:8:9: error: '_impure_ptr' is not a type
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: In file included from .\main.cpp:1:0:
[DEBUG] Output: .\main.cpp:8:9: error: expected identifier before '(' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: expected ')' before '->' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: expected ')' before '->' token
[DEBUG] Output: .\main.cpp:8:9: error: '_stdout' does not name a type
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: trailing return type only available with -std=c++11 or -std=gnu++11
[DEBUG] Output: .\main.cpp:8:9: error: expected ';' at end of member declaration
[DEBUG] Output: .\main.cpp:8:9: error: expected unqualified-id before ')' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
Traceback (most recent call last):
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py", line 290, in <module>
    ignore=options.ignore)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\build_api.py", line 586, in build_project
    objects = toolchain.compile_sources(resources, resources.inc_dirs)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\toolchains\__init__.py", line 916, in compile_sources
    return self.compile_seq(queue, objects)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\toolchains\__init__.py", line 930, in compile_seq
    res['command']
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\toolchains\__init__.py", line 1084, in compile_output
    raise ToolException(_stderr)
ToolException: In file included from c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\arm-none-eabi\include\stdio.h:60:0,
                 from .\main.cpp:1:
.\main.cpp:8:9: error: '_impure_ptr' is not a type
    void putchar(char c);
         ^
In file included from .\main.cpp:1:0:
.\main.cpp:8:9: error: expected identifier before '(' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
[... cut ...]
[mbed] Exec "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM -v" in C:\dev_temp\mbed-os-emb-cli-cryptic-build-err
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM -v" in "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err"
---
```

bmcdonnell-ionx, 2018-07-05:
@0xc0170,

```
me@pc MINGW64 /c/dev_temp/mbed-os-emb-cli-cryptic-build-err (master)
$ mbed compile -vv -m LPC4088
[Error] @,: Compiler version mismatch: Have 7.3.1; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-emb-cli-cryptic-build-err (LPC4088, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Macros: -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DMBED_BUILD_TIMESTAMP=1530825487.72 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4
Compile [100.0%]: main.cpp
Compile: C:/My Program Files/GNU Tools ARM Embedded/gcc-arm-none-eabi-7-2018-q2-update-win32/bin\arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -D_RTE_ -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DMBED_BUILD_TIMESTAMP=1530825487.72 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4 @.\BUILD\LPC4088\GCC_ARM\.includes_eec51d9479a32bdfc5a7b9e56c0c7248.txt -include .\BUILD\LPC4088\GCC_ARM\mbed_config.h -MD -MF BUILD\LPC4088\GCC_ARM\main.d -o BUILD\LPC4088\GCC_ARM\main.o .\main.cpp
[DEBUG] Return: 0
Link: mbed-os-emb-cli-cryptic-build-err
Preproc: C:/My Program Files/GNU Tools ARM Embedded/gcc-arm-none-eabi-7-2018-q2-update-win32/bin\arm-none-eabi-cpp -E -P .\mbed-os\targets\TARGET_NXP\TARGET_LPC408X\device\TOOLCHAIN_GCC_ARM\LPC4088.ld -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -o .\BUILD\LPC4088\GCC_ARM\.link_script.ld
[DEBUG] Return: 0
Link: C:/My Program Files/GNU Tools ARM Embedded/gcc-arm-none-eabi-7-2018-q2-update-win32/bin\arm-none-eabi-gcc @.\BUILD\LPC4088\GCC_ARM\.link_files.txt
[DEBUG] Return: 0
Elf2Bin: mbed-os-emb-cli-cryptic-build-err
FromELF: C:/My Program Files/GNU Tools ARM Embedded/gcc-arm-none-eabi-7-2018-q2-update-win32/bin\arm-none-eabi-objcopy -O binary .\BUILD\LPC4088\GCC_ARM\mbed-os-emb-cli-cryptic-build-err.elf .\BUILD\LPC4088\GCC_ARM\mbed-os-emb-cli-cryptic-build-err.bin
[DEBUG] Return: 0
[DEBUG] LPC Patch: mbed-os-emb-cli-cryptic-build-err.bin
+------------------+-------+-------+------+
| Module           | .text | .data | .bss |
+------------------+-------+-------+------+
| [fill]           |    80 |     0 |   13 |
| [lib]\c.a        | 18771 |  2472 |   89 |
| [lib]\gcc.a      |  3160 |     0 |    0 |
| [lib]\misc       |   208 |    12 |   28 |
| main.o           |     2 |     0 |    0 |
| mbed-os\drivers  |    56 |     0 |    0 |
| mbed-os\hal      |  1757 |     4 |   68 |
| mbed-os\platform |  2713 |   260 |  133 |
| mbed-os\rtos     |  9351 |   168 | 6073 |
| mbed-os\targets  |  3096 |    16 |   16 |
| Subtotals        | 39194 |  2932 | 6420 |
+------------------+-------+-------+------+
Total Static RAM memory (data + bss): 9352 bytes
Total Flash memory (text + data): 42126 bytes

Image: .\BUILD\LPC4088\GCC_ARM\mbed-os-emb-cli-cryptic-build-err.bin
[mbed] Working path "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err" (program)
[mbed] Query "c:\python27\python.exe -m pip list -l" in C:\dev_temp\mbed-os-emb-cli-cryptic-build-err
Package          Version
---------------- -----------
appdirs          1.4.3
beautifulsoup4   4.6.0
certifi          2017.11.5
chardet          3.0.4
colorama         0.3.9
Cython           0.27.3
enum34           1.1.6
fasteners        0.14.1
funcsigs         1.0.2
functools32      3.2.3.post2
future           0.16.0
fuzzywuzzy       0.15.1
idna             2.6
intelhex         2.1
Jinja2           2.10
jsonschema       2.6.0
junit-xml        1.8
lockfile         0.12.2
MarkupSafe       1.0
mbed-cli         1.7.3
mbed-greentea    1.3.2
mbed-host-tests  1.2.0
mbed-ls          1.3.3
mercurial        4.6
mock             2.0.0
monotonic        1.4
numpy            1.14.0
pbr              3.1.1
pip              10.0.1
prettytable      0.7.2
pyelftools       0.24
pyOCD            0.8.1a1
pyserial         3.4
pywinusb         0.4.2
PyYAML           3.12
requests         2.18.4
setuptools       28.8.0
six              1.11.0
urllib3          1.22
websocket-client 0.44.0
[mbed] Exec "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM -v" in C:\dev_temp\mbed-os-emb-cli-cryptic-build-err
```

deepikabhavnani, 2018-07-05:
>[DEBUG] Output: .\main.cpp:8:9: error: trailing return type only available with -std=c++11 or -std=gnu++11

Can you try updating C++ flags to gnu++11 in https://github.com/ARMmbed/mbed-os/blob/master/tools/profiles/develop.json#L11 and see if it builds with 6-2017-q2-update

bmcdonnell-ionx, 2018-07-05:
@deepikabhavnani,

Same result w/ gnu++11.

```
me@pc MINGW64 /c/dev_temp/mbed-os-emb-cli-cryptic-build-err (master)
$ mbed compile -vv -m LPC4088
Building project mbed-os-emb-cli-cryptic-build-err (LPC4088, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Scan: FEATURE_LWIP
Macros: -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1530827069.85 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4
Compile [  0.3%]: main.cpp
Compile: arm-none-eabi-g++ -std=gnu++11 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -D_RTE_ -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DDEVICE_EMAC=1 -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_DEBUG_AWARENESS=1 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NXP_EMAC -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_LPCTarget -DTARGET_LPC408X -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_MCU_LPC4088 -D__FPU_PRESENT=1 -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_LPC4088 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_BUILD_TIMESTAMP=1530827069.85 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4 @.\BUILD\LPC4088\GCC_ARM\.includes_eec51d9479a32bdfc5a7b9e56c0c7248.txt -include .\BUILD\LPC4088\GCC_ARM\mbed_config.h -MD -MF BUILD\LPC4088\GCC_ARM\main.d -o BUILD\LPC4088\GCC_ARM\main.o .\main.cpp
[Error] main.cpp@8,9: '_impure_ptr' is not a type
[Error] main.cpp@8,9: expected identifier before '(' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: expected ')' before '->' token
[Error] main.cpp@8,9: '_stdout' does not name a type
[Error] main.cpp@8,9: '__sputc_r' function with trailing return type not declared with 'auto' type specifier
[Error] main.cpp@8,9: expected ';' at end of member declaration
[Error] main.cpp@8,9: expected unqualified-id before ')' token
[DEBUG] Return: 1
[DEBUG] Output: In file included from c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\arm-none-eabi\include\stdio.h:60:0,
[DEBUG] Output:                  from .\main.cpp:1:
[DEBUG] Output: .\main.cpp:8:9: error: '_impure_ptr' is not a type
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: In file included from .\main.cpp:1:0:
[DEBUG] Output: .\main.cpp:8:9: error: expected identifier before '(' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: expected ')' before '->' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: expected ')' before '->' token
[DEBUG] Output: .\main.cpp:8:9: error: '_stdout' does not name a type
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
[DEBUG] Output: .\main.cpp:8:9: error: '__sputc_r' function with trailing return type not declared with 'auto' type specifier
[DEBUG] Output: .\main.cpp:8:9: error: expected ';' at end of member declaration
[DEBUG] Output: .\main.cpp:8:9: error: expected unqualified-id before ')' token
[DEBUG] Output:     void putchar(char c);
[DEBUG] Output:          ^
Traceback (most recent call last):
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py", line 290, in <module>
    ignore=options.ignore)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\build_api.py", line 586, in build_project
    objects = toolchain.compile_sources(resources, resources.inc_dirs)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\toolchains\__init__.py", line 914, in compile_sources
    return self.compile_queue(queue, objects)
  File "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\toolchains\__init__.py", line 977, in compile_queue
    raise ToolException(err)
ToolException: In file included from c:\program files (x86)\gnu tools arm embedded\6 2017-q2-update\arm-none-eabi\include\stdio.h:60:0,
                 from .\main.cpp:1:
.\main.cpp:8:9: error: '_impure_ptr' is not a type
    void putchar(char c);
         ^
In file included from .\main.cpp:1:0:
.\main.cpp:8:9: error: expected identifier before '(' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
    void putchar(char c);
         ^
.\main.cpp:8:9: error: expected ')' before '->' token
.\main.cpp:8:9: error: '_stdout' does not name a type
    void putchar(char c);
         ^
[... cut ...]
[mbed] Exec "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM -v" in C:\dev_temp\mbed-os-emb-cli-cryptic-build-err
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\dev_temp\mbed-os-emb-cli-cryptic-build-err\mbed-os\tools\make.py -t GCC_ARM -m LPC4088 --source . --build .\BUILD\LPC4088\GCC_ARM -v" in "C:\dev_temp\mbed-os-emb-cli-cryptic-build-err"
---
```

0xc0170, 2018-07-20:
`putchar` is a macro isn't it, thus in your main.cpp file it gets overwritten by a macro to something that is not valid in C++ class? Please check preprocessed file how the class looks like

bmcdonnell-ionx, 2018-07-20:
> `putchar` is a macro isn't it

Not necessarily, but apparently [it can be in C](https://stackoverflow.com/a/16165900). Is this also true in C++?

If so, cryptic-ness of the failure notwithstanding, does that mean this failure to compile is "acceptable" behavior? Such that creating a member function by the same name as an `stdio` function is not "guaranteed" to work? Would that hold true for both `<stdio.h>` and `<cstdio>`?

0xc0170, 2018-07-23:
My assumption was made based on the errors - obviously something is overwritten class method and adding there some additional text that is causing it fail to compile (impure pointer is one of them) - you can verify after preprocessor stage - what is the definiton for putchar there. 

> Such that creating a member function by the same name as an stdio function is not "guaranteed" to work?

I haven't seen this before but would expect not to use it as it can be macro and that can cause problems. If you look at cstdio header file, you will find there `undef putchar`

bmcdonnell-ionx, 2018-07-23:
It does compile w/ GCC_ARM 6 when I `#include <cstdio>` instead of `<stdio.h>`. 

Should I close this issue, then?

closed 2018-07-23 by 0xc0170