ARMmbed/mbed-os #7416: HardFault when sending data using TCP socket in a Thread

flipflop22, 2018-07-04:
### Description

My setup:
-UBLOX_EVK_ODIN_W2
-GCC_ARM
-mbed cli (newest version)
-GNU ARM Eclipse
-Issue: networking in thread

So I try to create a thread.
`send_thread(osPriorityRealtime)`

I try start the thread.

I create a buffer with size of 5360
`			for(i=0;i<5360;i++){
				out_buffer[i]=0xAA;
			}`

when I try to send this buffer.
`size = client_sock.send(out_buffer,BUFFER_SIZE);`

Then, Mbed Error message popped out says HardFault as follow:

> Context:
> R0   : AAAAAAAA
> R1   : 000000AA
> R2   : DC02A8C0
> R3   : 00000000
> R4   : 2000FCB8
> R5   : 2001023C
> R6   : 00000860
> R7   : 00000000
> R8   : 00000001
> R9   : 2000FCBC
> R10  : 00000000
> R11  : 20000EC4
> R12  : 0801DE0D
> SP   : 2001CE40
> LR   : 0800943F
> PC   : 0800593C
> xPSR : 610F0000
> PSP  : 2001CDD8
> MSP  : 2002FFD8
> CPUID: 410FC241
> HFSR : 40000000
> MMFSR: 00000000
> BFSR : 00000082
> UFSR : 00000000
> DFSR : 0000000A
> AFSR : 00000000
> SHCSR: 00000000
> BFAR : AAAAAAEB
> Mode : Thread
> Priv : Privileged
> Stack: PSP
> 
> Threads Info:
> Current:
> State: 0x00000002 Entry: 0x0801F2DD Stack Size: 0x00001000 Mem: 0x2001BF78 SP: 0x2001CE58
> Next:
> State: 0x00000002 Entry: 0x0801F2DD Stack Size: 0x00001000 Mem: 0x2001BF78 SP: 0x2001CE58
> Wait:
> State: 0x00000083 Entry: 0x0801F211 Stack Size: 0x00000300 Mem: 0x20005770 SP: 0x20005A08
> Delay:
> State: 0x00000083 Entry: 0x0801F2DD Stack Size: 0x00001000 Mem: 0x20016BD0 SP: 0x20017A98
> Idle:
> State: 0x00000001 Entry: 0x0801C2F1 Stack Size: 0x00000200 Mem: 0x20005A70 SP: 0x20005C18
> 
> -- MbedOS Fault Handler --
> 
> 
> 
> ++ MbedOS Error Info ++
> Error Status: 0x80ff013d Code: 317 Module: 255
> Error Message: System encountered an unrecoverable fault exception, halting system.
> Location: 0x800593c
> 
> -- MbedOS Error Info --

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

kjbracey-arm, 2018-07-04:
Are you sure you've not just overrun the buffer when filling it? The system has tried to use 0xAAAAAAAA as an address. Is the buffer on the stack? It wouldn't fit.

SeppoTakalo, 2018-07-23:
Needs example code to replicate.

flipflop22, 2018-07-25:
problem solved. The problem is with the code. Apparently the input size is bigger than declared buffer.

closed 2018-07-25 by flipflop22