ARMmbed/mbed-os #7418: Compiler cache ccache sccache

drahnr, 2018-07-04:
### Description

How to integrate sccache with mbed os? Setting `GCC_ARM_PATH` does not suffice since I need an additional compiler profile then which is essentailly a hacked `GCC_ARM` to provider the additional pseudo option `arm-none-...gcc`

### Issue request type
[x] Question  
[ ] Enhancement  
[ ] Bug

theotherjimmy, 2018-07-05:
@drahnr You just described the only way I know how to do it. You're requesting first-class support fo sccache then?

drahnr, 2018-07-05:
Yes, because with multiple targets recompiling and testing becomes time consuming.

drahnr, 2018-07-06:
An idea would be to use an env var like `MBED_COMPILED_OBJECT_CACHE_BINARY` and then just prefix execution with that.

theotherjimmy, 2018-07-06:
Be aware that this is low on my list of things to do. I'll get around to it sometime, and the best way to have it faster is to submit a PR that implements it.

drahnr, 2018-07-06:
Are you ok with the env var approach or should this be done differently? If not, I'd look into `mbed-cli` and how to get that into it.

theotherjimmy, 2018-07-06:
@drahnr, it should be configurable from both an env var and `mbed config <var>` (which uses env vars).

closed 2018-07-24 by theotherjimmy