ARMmbed/mbed-os #7422: Link error: Symbol __user_perthread_libspace multiply defined

toyowata, 2018-07-05:
### Description

I got link error below when I build mbed-os-example-blinky project for any target.
```
Error: Symbol __user_perthread_libspace multiply defined (by /build/mbed-os.lib/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.GR_LYCHEE.o and /build/mbed-os.lib/rtos/TARGET_CORTEX/mbed_boot.GR_LYCHEE.o).
Error: Not enough information to list the image map.
Info: Unable to download. Fix the reported errors...
```

[Steps to reproduce]

* Import [mbed-os-example-blinky](https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-blinky/) program
* Set target to Mbed OS 5 support platform (e.g. Renesas GR-LYCHEE, NXP K64F etc)
* Compile the program

[Other notes]

* This link error cannot be shown in command line build by Mbed-CLI

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

BryceeeeeChen, 2018-07-05:
I got same issue for L476RG....
Error: Symbol __user_perthread_libspace multiply defined (by /build/mbed-os.lib/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.NUCLEO_L476RG.o and /build/mbed-os.lib/rtos/TARGET_CORTEX/mbed_boot.NUCLEO_L476RG.o).

jeromecoutant, 2018-07-05:
I confirmed the error with NUCLEO_F746ZG and only ARM toolchain with 5.9.2 label.

[DEBUG] Errors: Error: L6200E: Symbol __user_perthread_libspace multiply defined (by BUILD/tests/NUCLEO_F746ZG/ARM/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.o and BUILD/tests/NUCLEO_F746ZG/ARM/rtos/TARGET_CORTEX/mbed_boot.o).

jeromecoutant, 2018-07-05:
Hi again
I have compared my local profile with the one in tools/profiles,
and it seems that -D_RTE_ was missing.
I can't explain what it is, but it solves my compilation issue...

theotherjimmy, 2018-07-05:
Thanks @jeromecoutant I'll look into it.

0xc0170, 2018-07-09:
@jeromecoutant RTE is defined for 5.9.2 . is that causing a problem? 

cc @ARMmbed/mbed-os-core

0xc0170, 2018-07-09:
I suspect this is the one that is causing a problem #6973. I'll check why rtx lib does not see the no multithreaded macro and why this was not seen in our CI previously

0xc0170, 2018-07-09:
@theotherjimmy I referenced 7447 but that one is about own profile. This issue is in the online compiler. Can you check that profiles were updated there? RTE symbol defined? (until 5.9.3 contains proper fix for this)

theotherjimmy, 2018-07-09:
See https://github.com/ARMmbed/mbed-os/pull/7449

theotherjimmy, 2018-07-09:
We need to backport that patch into 5.9.2 to fix the problem in the online compiler.

mharmouche, 2018-07-11:
I'm blocked since one week due to this error.
Any solution?

cmonr, 2018-07-11:
@mharmouche The PR (https://github.com/ARMmbed/mbed-os/pull/7474) has just been merged and is now on master. Would you mind retesting?

cmonr, 2018-07-11:
@toyowata Would you mind confirming that the issue has been fixed?

mharmouche, 2018-07-11:
Issue not fixed!

mharmouche, 2018-07-11:
[image]

mecwerks, 2018-07-11:
@cmonr I can confirm the newest commit works. Although I couldn't find a way to update the official library to the newest commit. So I resorted to forking it, and pushing my own release tag to make sure it was using the newest. Compiled and ran for me.

cmonr, 2018-07-11:
@mecwerks Correct. This is by design with mbed-cli.

Whenever I need to test changes that haven't hit a release, I go into the mbed-os directory and do a manual git checkout.

cmonr, 2018-07-11:
> I'm blocked since one week due to this error.

@mharmouche This is the first time that I've read that you're having an issue. Did you open one elsewhere?
Cna you provide output of the mbed-os version that you're running?

0xc0170, 2018-07-12:
The master branch contains the fix. The next release 5.9.3 will include the fix.

Meanwhile, to fix this issue - all we need is to have _RTE_ defined in the online compiler. Please add mbed_lib.json file with the following (if you already have the config, just add macro):

```
{
    "name": "rtos_fix",
    "macros": ["_RTE_"]
}

```

Tested, works 👍

toyowata, 2018-07-13:
@0xc0170 Thank you very much for your testing.
I also confirmed that adding compiler macros fixes the problem.

* Add ```_RTE_``` macros from [Compile] - [Compile macros]
[image]

* Then, [Compile] - [Compile All]

cmonr, 2018-07-13:
Closing since issue appears resolved.

diegomhamilton, 2018-12-15:
I'm getting this error now and adding the _RTE_ macro didn't solve it.

cmonr, 2018-12-16:
@diegomhamilton Please open a new issue referencing this issue. We prefer to see new issues come in instead of reviving old ones.