ARMmbed/mbed-os #7425: I2C driver not initialized when writing before reading

grathlinkbit, 2018-07-05:
### Description

**Target:** NRF52_DK
Firmware: 0246_sam3u2c_mkit_dk_dongle_nrf5x_0x5000.bin [link](https://armmbed.github.io/DAPLink)
**Toolchain:** GCC_ARM (newest version)
**Tools:** Mbed-cli, exported to Microsoft Visual Code for debugging
(command used: mbed export -i vscode_gcc_arm -m NRF52_DK)
[develpment-board ](https://os.mbed.com/platforms/Nordic-nRF52-DK/)
[I2C/TWI device](https://www.nxp.com/docs/en/data-sheet/PCF8523.pdf)
Operating System: Ubuntu 18.10
**SHA:** [mbed-os-5.9.2](https://github.com/ARMmbed/mbed-os/#62f8b922b420626514fd4690107aff4188469833)

**Steps to reproduce:**
Connect the pcf8523 to the DK.
run this code:
```
#include <mbed.h>
int main()
{

        I2C i2c_obj(I2C_SDA0, I2C_SCL0);
        i2c_obj.frequency(100000);
        uint8_t addr = 0x00;
        uint8_t* addr_ptr = &addr;
        uint8_t reg_contr_one = 0;
        uint8_t* reg_ptr = &reg_contr_one;
        /* writing */
        PRINT("Writing started\r\n");
        bool bWorked = false;
        i2c_obj.start();
        bWorked = i2c_obj.write(0x68 << 1);
        if(!bWorked)
        {
            PRINT("write failed : \r\n");
        }
        bWorked = false;
        bWorked = i2c_obj.write(0x00);
        if(!bWorked)
        {
            PRINT("write failed : \r\n");
        }
        bWorked = false;
        bWorked = i2c_obj.write(0x00);
        if(!bWorked)
        {
            PRINT("write failed : \r\n");
        }
        i2c_obj.stop();
        PRINT("Writing done\r\n");
        /* reading */
        PRINT("Reading started\r\n");
        bWorked = i2c_obj.write(PCF8523_ADDRESS << 1, (char*) addr_ptr, 1, true);
        if(!bWorked)
        {
            PRINT("write failed : \r\n");
        }
        bWorked = i2c_obj.read(PCF8523_ADDRESS << 1, (char*) reg_ptr, 1);
        if(!bWorked)
        {
            PRINT("write failed : \r\n");
        }
    return 0;
}
```
This will leave p_cb->state uninitialized and
` ASSERT(p_cb->state != NRF_DRV_STATE_UNINITIALIZED);` (nrf_drv_twi.c, line: 397)
 will trigger.

If a read operation is executed before the write command then this error will not occur. 
### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug

marcuschangarm, 2018-07-05:
Looks like the I2C equivalent of this bug: https://github.com/ARMmbed/mbed-os/pull/7280

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTDEV-1567*

naveenkaje, 2018-10-04:
Hi @grathlinkbit ,
I don't have access to the particular RTC chip you are using. But I tried with 2 I2C Slaves
https://os.mbed.com/teams/Seeed/code/Seeed_Barometer_Sensor_Example/ (Temp and Pressure Sensor)
and
https://os.mbed.com/teams/mbed/code/I2C_HelloWorld_Mbed/ (Using mbed CI Shileld that has a temperature sensor).

I have attached captured traffic that I have observed with above programs with mbed 5.9.7

[image]
[image]

I am wondering if this is issue specific to the RTC slave that you are using. Do you have access to an analyzer to debug further?

linlingao, 2019-05-30:
@grathlinkbit Do you still need help from us? Naveen asked for debug data 7 months ago. I'll go ahead and close this due to inactivity. Please reopen if the issue is not solved.

closed 2019-05-30 by linlingao