ARMmbed/mbed-os #7428: LoRaWAN Stack No Longer Builds with -DLORAWAN_COMPLIANCE_TEST

mattbrown015, 2018-07-05:
### Description
I don't understand the purpose of `-DLORAWAN_COMPLIANCE_TEST` but believe I probably should given that I'm writing software for a LoRaWAN device!

In an attempt to gain some understanding I defined `LORAWAN_COMPLIANCE_TEST` but there were compile errors in LoRaWANStack.cpp.  I fixed the first couple but more appeared and hence I think someone knowledgeable needs to have a look. :-}

I only started rummaging around in this code today but what I've seen so far leads me to believe this code could be very useful.  For example, I'm hoping it will make continuous Tx possible so we can test our antenna.

### Issue request type
[ ] Question  
[ ] Enhancement  
[X] Bug

mattbrown015, 2018-07-05:
@hasnainvirk I'm guessing at least some of the problems relate to [LoRa: State machine work #6808](https://github.com/ARMmbed/mbed-os/pull/6808).

Any help greatly appreciated! :-)

hasnainvirk, 2018-07-06:
@mattbrown015 This is known. Compliance test code port was in for legacy compliance support. The code is still in as we have plans to setup our own testing framework.

mattbrown015, 2018-07-06:
@hasnainvirk Ah, OK.

So perhaps you can offer some advice and help me understand whether or not this functionality would be anything like what I need?

We've built a board and the hardware team want to test the antenna.  They asked me to make the board transmit _continuously_ using a fixed SF and power.  They said _continuously_ but I'm not sure they meant that literally, I think they just meant _transmit LoRa packets a lot more often than normal_.  As I understand it, they don't care what's in the LoRa packets and whether or not it is connecting to a LoRaWAN gateway.

Some of what I saw in `LORAWAN_COMPLIANCE_TEST` made me think it might do what I want.  It appeared to do the LoRaWAN thing but accepted _special_ messages on port 224 that could select special modes.  I didn't spend too long staring at the code because I thought I'd just try it!

If `LORAWAN_COMPLIANCE_TEST` is going nowhere for me what other options do you think I have?

My initial thought was that I'd have to write a special test app that just used the radio drivers to talk to the SX1276 and make it transmit any old rubbish.  Naturally I need to give the hardware team what they want in the next day or 2.

Thanks!

mattbrown015, 2018-07-06:
@hasnainvirk Thanks for the advice, I shall digest it and talk to the hardware team. :-)

closed 2018-07-06 by mattbrown015

mattbrown015, 2018-09-21:
(For completeness just in case someone stumbles across this issue.)

I notice some work has been done on this now, [LoRaWAN: Refactoring compliance test code #8074](https://github.com/ARMmbed/mbed-os/pull/8074).