ARMmbed/mbed-os #7432: LoRaWAN maximum payload length over-enforced at low data rates

EnsilicaEdd, 2018-07-06:
### Description
I've been doing some testing at low data rates with the LoRaWAN stack, and have been receiving failure code -1010 from LoRaWANInterface::send().

This is LORAWAN_STATUS_LENGTH_ERROR, and it seems that the stack thinks my message is too long.  However, it's 50 bytes, which should be fine, since the maximum payload length at low data rates (for EU868 at least) is 51 bytes.

It seems that LoRaMac::prepare_frame is adding the LoRaWAN header to the message (an extra 13 bytes) before checking the length, and hence failing.  Unless I've misunderstood something, the check should be _before_ the header is added.

I'm almost up-to-date ( SHA: 9f27672f0 ), and it doesn't look like this has been touched by any recent changes.

### Issue request type
    [ ] Question
    [ ] Enhancement 
    [X] Bug

kivaisan, 2018-07-09:
@EnsilicaEdd, thanks for the report. I just checked the spec and indeed the maximum length should be "The maximum application payload length in the absence of the optional FOpt control field". We'll check this.

EnsilicaEdd, 2018-07-20:
Just to say: I'm not ignoring the fix, I'm just not in a position to verify it right now.  Will hopefully get a moment in the next week or so.

EnsilicaEdd, 2018-07-31:
I've just sent a 48-byte message at DR0 / SF12.  So that's working as expected.  Thanks!

closed 2018-07-31 by EnsilicaEdd