ARMmbed/mbed-os #7433: turning on optimization for mbed-os only

drahnr, 2018-07-06:
### Description
So it turns out that the mbed-os optimization has significant overhead (GPIO, Timer) compared to bare metal such that our usecase is impossible to do when compiling with `-ggdb -O2`.

Is there a rather painfree way to compile mbed-os itself with a different set of compiler options than the rest of the project? After all we still need to debug our end of the stack.

### Issue request type

[X] Question  
[ ] Enhancement  
[ ] Bug

AGlass0fMilk, 2018-07-10:
If you're using mbed-cli, you may be able to use the --library flag to break your compilation into steps (allowing you to select different build profiles for mbed-os and the rest of your code):

https://github.com/ARMmbed/mbed-cli#compiling-static-libraries

theotherjimmy, 2018-07-17:
@drahnr You'll want to use --no-archive or similar option, as some weak symbols break on arm compiler.

drahnr, 2018-07-17:
@AGlass0fMilk @theotherjimmy I am running in plenty of undefined symbols (even with `--no-archive --lib`) when trying to cerate a twostage build.

drahnr, 2018-07-17:
mbed-os 5.7
```
[Error] reloc_vector_table.c@85,46: 'MBR_VTOR_ADDRESS' undeclared (first use in this function)
[Error] reloc_vector_table.c@89,53: 'SOFTDEVICE_VTOR_ADDRESS' undeclared (first use in this function); did you mean 'SOFTDEVICE_INFO_STRUCT_ADDRESS'?
[Warning] reloc_vector_table.c@98,5: implicit declaration of function 'nrf_dfu_mbr_init_sd' [-Wimplicit-function-declaration]
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c: In function 'nrf_reloc_vector_table':
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c:85:46: error: 'MBR_VTOR_ADDRESS' undeclared (first use in this function)
     uint32_t *mbr_vtor_address = (uint32_t *)MBR_VTOR_ADDRESS;
                                              ^~~~~~~~~~~~~~~~
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c:85:46: note: each undeclared identifier is reported only once for each function it appears in
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c:89:53: error: 'SOFTDEVICE_VTOR_ADDRESS' undeclared (first use in this function); did you mean 'SOFTDEVICE_INFO_STRUCT_ADDRESS'?
     uint32_t *softdevice_vtor_address = (uint32_t *)SOFTDEVICE_VTOR_ADDRESS;
                                                     ^~~~~~~~~~~~~~~~~~~~~~~
                                                     SOFTDEVICE_INFO_STRUCT_ADDRESS
mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c:98:5: warning: implicit declaration of function 'nrf_dfu_mbr_init_sd' [-Wimplicit-function-declaration]
     nrf_dfu_mbr_init_sd ();
     ^~~~~~~~~~~~~~~~~~~

Completed in: (9.32)s

Build failures:
  * GCC_ARM::NELLO_ONE_HW_3_3
[mbed] ERROR: "/usr/bin/python2" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python2 -u somewhere/mbed-os/tools/build.py -t GCC_ARM -m NELLO_ONE_HW_3_3 --profile ./config/debug.json --source ./scratchpad --source mbed-os --build ../mbed-os-build --no-archive" in "somewhere"

```

theotherjimmy, 2018-07-17:
@drahnr You're missing the destination of `--build`, but assuming you have one, that command should work.

Other than that, I don't know what could be going wrong. You could look at the macros in `mbed_config.h` and on the command line to see if it's missing something important.

drahnr, 2018-07-18:
@theotherjimmy  Yep that was a copy paste mistake on my end, but it already fails with the first command. The `mbed_config.h` only inlcudes the config values of my application but but nothing regarding the softdevice.

When I add a `-v` to the pool, I see that the `-DS132` is in there which is causing the issue.

```
Compile [100.0%]: reloc_vector_table.c
[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g3 -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DTARGET_MCU_NRF52832 -DTARGET_SDK11 -D__MBED__=1 -DNRF52_PAN_53 -DTARGET_NRF52_COMMON -DNRF52_PAN_36 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_PORTINOUT=1 -DBOARD_PCA10040 -DDEVICE_ITM=1 -DTARGET_FF_ARDUINO -DNRF52_PAN_12 -DTOOLCHAIN_GCC -DNRF52_PAN_58 -DDEVICE_SPISLAVE=1 -DNRF52_PAN_55 -DTARGET_NELLO_ONE -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_NRF52_DK -DTARGET_NELLO_ONE_BREADBOARD -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_NRF5 -DFEATURE_BLE=1 -DTARGET_MCU_NRF52 -DCMSIS_VECTAB_VIRTUAL -DMBED_BUILD_TIMESTAMP=1531927841.63 -DDEVICE_SPI_ASYNCH=1 -DMBED_TICKLESS -DNRF52 -DTARGET_CORTEX -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NRF52832 -D__CORTEX_M4 -DDEVICE_SERIAL=1 -D__FPU_PRESENT=1 -DNRF52_PAN_63 -DDEVICE_SLEEP=1 -DDEVICE_PORTIN=1 -DNRF52_PAN_51 -DDEVICE_STCLK_OFF_DURING_SLEEP=1 -DTARGET_NORDIC -DNRF52_PAN_30 -DBLE_STACK_SUPPORT_REQD -DNRF52_PAN_15 -DCONFIG_GPIO_AS_PINRESET -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DNRF52_PAN_31 -DTOOLCHAIN_GCC_ARM -DSWI_DISABLE0 -DSOFTDEVICE_PRESENT -DNRF52_PAN_20 -DDEVICE_I2C_ASYNCH=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DDEVICE_INTERRUPTIN=1 -DNRF52_PAN_64 -DTARGET_NELLO_MONO_LED -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DNRF52_PAN_62 -DNRF52_PAN_54 -DDEVICE_SPI=1 -DDEVICE_FLASH=1 -DS132 -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 @../BUILD_MBED_OS/.includes_b848167550f84002cc5bdf2b9c66e187.txt -include ../BUILD_MBED_OS/mbed_config.h -MD -MF ../BUILD_MBED_OS/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.d -o ../BUILD_MBED_OS/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.o mbed-os/targets/TARGET_NORDIC/TARGET_NRF5/reloc_vector_table.c
```
which implies that my `app_config.json` is not read completely.

AGlass0fMilk, 2018-07-18:
I noticed your custom target inherits from `NRF52_DK`

I had issues when doing this as well, since the recent SDK14.2 update introduced an mbed_lib.json for the NRF52 targets. Inadvertently, some macros/features get added that you wouldn't expect since you are inheriting from the NRF52_DK. See comments on this issue #6824

This may not be the cause of your problem but it may cause other issues in the future. The documentation was updated to include info on creating a custom target here: https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_NORDIC/TARGET_NRF5x#adding-new-targets-based-on-nordic-nrf52832-and-nrf52840-mcus

Maybe some part of the configuration is taking precedent over you overrides? I'm not sure how mbed-cli treats mbed_app.json configurations when building with the --library flag.

drahnr, 2018-07-18:
@theotherjimmy that's already given above and it does not work

theotherjimmy, 2018-07-18:
@drahnr When you say "it does not work" that can mean many things:
 * It does not accept the option.
 * Nothing in the app config has an affect.
 * Part of the app config have an affect, but others don't.

Could you be more specific?

theotherjimmy, 2018-07-19:
Oh, yep. They simply don't take in an app config  XD sorry about that. It looks like Mbed CLI eats the parameter and does nothing with it. https://github.com/ARMmbed/mbed-os/blob/master/tools/build.py#L191-L201 Invocation of build_library does not have it on master. If you only care about removing features/defines, then custom target or a lib config could get you the same thing.

drahnr, 2018-07-19:
Actually, `custom_targets.json` doesn't, they never get applied - that's why I moved them into the app_config.json in the first place...
A lib config seems to be the wrong place to store this.

theotherjimmy, 2018-07-20:
@drahnr You're right:

>A lib config seems to be the wrong place to store this.

But if it's your only remaining option, you should at least try it.

theotherjimmy, 2018-07-20:
Also note that `custom_target.json` defines new targets, and does not override targets defined in `target.json`.

drahnr, 2018-07-20:
@theotherjimmy I figured that out the hard way (there is another ticket I filed a few months back). This is becoming increasingly crude when all that I want is to make a multi target project build faster

theotherjimmy, 2018-07-23:
@drahnr Yeah

> I figured that out the hard way

I'd like to make that not happen for others. Where did you look in our docs for adding custom targets? How could I make it more clear that they do not override?

> This is becoming increasingly crude when all that I want is to make an ISR fast enough for my purposes 

I can fix that, expecting that your task is a common one. OTOH, I can't (yet) change the tools you work with on 5.7.

OTOH, you could create a custom build profile that makes use of most of the debug profile, but changes the amount of the debug information present. Check out the `tools/profiles` directory for the built-in profiles.

drahnr, 2018-07-24:
I can not migrate until https://github.com/ARMmbed/mbed-os/issues/7463 is fixed.

theotherjimmy, 2018-07-24:
Good to know. I'll make the changes needed.

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-238*

linlingao, 2019-06-12:
@drahnr Have you received/verified changes from @theotherjimmy ? Can we close this?

closed 2019-10-31 by ciarmcom