ARMmbed/mbed-os #7438: Device-key failing unit test on K64 / IAR

Alex-EEE, 2018-07-06:
Numerous errors with device-key on mbed test, but for instance:

| K64F-IAR | K64F          | mbed-mbed-os-features-device_key-tests-device_key-functionality | TIMEOUT | 58.8

K64F-IAR | K64F          | mbed-mbed-os-features-device_key-tests-device_key-functionality | Device Key - long consistency test                                                     | 0      | 0      | ERROR   | 0.0

### Issue request type
    [] Question
    [ ] Enhancement 
    [X] Bug 
-->

0xc0170, 2018-07-09:
What is IAR version, Mbed OS version/commit sha ?

0xc0170, 2018-07-09:
I reviewed some recent PR, did not spot any failure for these tests, target and toolchain.

yossi2le, 2018-07-09:
@Alex-EEE 
I am trying to reproduce the issue but in my case, it runs with no problems and there is no timeout error.
I am testing it with IAR compiler version V7.80.3.12102/W32. and mbed-os mbed-os-5.9.2.
Can you please elaborate with more details and reproducing flow.

Alex-EEE, 2018-07-09:
Ok, I'm on IAR 8.3, that might be the difference.  7.8 is well over a year old now, are there any plans to upgrade the supported version of IAR?

Thanks

yossi2le, 2018-07-10:
I have looked at the documentation for mbed os 5.9 and right now we are supporting IAR version 7.8. Can you please try to compile your code on this version and let us know if the problem is solved.
Unfortunately, I have no info regarding when the supported version will change.

0xc0170, 2018-07-10:
> Ok, I'm on IAR 8.3, that might be the difference. 7.8 is well over a year old now, are there any plans to upgrade the supported version of IAR?

We would like to update, there is no ETA. There is feature branch for IAR 8 update. We noticed some breaking changes there so be aware.

Alex-EEE, 2018-07-10:
@0xc0170 can you elaborate some on the issues with compiling mbedOS with IAR 8?  I looked at the branch and mainly noticed PRs around tweaks to support specific targets.  Is there a known "big issue" that we might be stepping into?

dannybenor, 2018-08-21:
@Alex-EEE can you close this issue? Will be solved when mbedos will move to IAR 8.3

Alex-EEE, 2018-08-21:
@dannybenor any ETA on that move?

closed 2018-08-21 by Alex-EEE

dannybenor, 2018-08-21:
@Alex-EEE Currently don't have info