ARMmbed/mbed-os #7439: Failing some filesystem tests for a specific tool-chain

aashishc1988, 2018-07-06:
After a merge from feature-emac to master, we are failing some filesystem tests for gcc_arm toolchain (6-2017-q2-update). The platform I am running these tests on is KL25Z.

The sha1 for the commit is
13dcef63e31ce972dbca1f306175d92a2a62d5e2

Tests we are failing:

features-tests-filesystem-util_block_device
features-tests-filesystem-mbr_block_device 
features-tests-filesystem-heap_block_device
features-tests-filesystem-buffered_block_device  (although i think this test was added as a part of that merge).

Steps to reproduce:

1. Checkout the sha1 I indicated above.
2. Compile
3. mbedgt -g KL25Z:raas_client:1x.xxx.xx.xx:xxxx -n features-tests-filesystem-util_block_device -v -V
(fill in the right ip and port number when running the test)

I cannot reproduce these failures with other toolchains (ARM, IAR).

    [ ] Question
    [ ] Enhancement 
    [X ] Bug

aashishc1988, 2018-07-11:
@davidsaada , I checked the PR and we are now passing all those tests. :)
Also, the test tests-mbed_platform-error_handling was passing even before the fix, so I am not sure if that raises more concerns or not.

davidsaada, 2018-08-01:
This can be closed now that [#7465](https://github.com/ARMmbed/mbed-os/pull/7465) was merged.

aashishc1988, 2018-08-01:
I pulled in the merge and verified the fix again. Closing this.

closed 2018-08-01 by aashishc1988