ARMmbed/mbed-os #7440: nRF52 BLE Connection Timeout

AGlass0fMilk, 2018-07-06:
I recently synced my mbed-os fork with master (as of now) and I can no longer connect to my application using either nRF Connect (Desktop, with nRF51 dongle as the BLE interface), or the Punch Through Bluetooth app on my iPhone.

Assuming my app had a bug, I went to this [repo](https://github.com/ARMmbed/mbed-os-example-ble) and compiled several of the examples after running "mbed update" (which presumably pulled to mbed-os-5.9.2). I am running on an NRF52_DK using the GCC_ARM toolchain (building with mbed-cli).

I am experiencing the same issue with the examples I tried (notably, the BLE_BatteryLevel example).

Can anyone else confirm that they can't connect to the BatteryLevel example on the NRF52_DK with mbed-os-5.9.2? I'm trying to get up to date so I can start developing with the new SecurityManager API.

I am running
[ ] Question  
[ ] Enhancement  
[ X ] Bug

AGlass0fMilk, 2018-07-07:
I have built the BLE_BatteryLevel demo by checking out the mbed-os.5.9.1 release and it worked as expected.

Something must have changed between 5.9.1 and 5.9.2 that is breaking connection establishment.

AGlass0fMilk, 2018-07-07:
I think I found the issue. Something in serial_api.c is causing the BLE stack to become unresponsive it seems.

I reverted serial_api.c to the mbed-os-5.9.1 release version and the example works as expected.

Any thoughts @marcuschangarm?

marcuschangarm, 2018-07-07:
I see the problem. We've been mislabeling the reserved SWI. It is not just SWI1 that's been reserved for the SoftDevice but SWI1, SWI2, SWI4, and SWI5.

marcuschangarm, 2018-07-09:
@AGlass0fMilk 

Can you give this PR a go and see if it solves the problem, please?

https://github.com/ARMmbed/mbed-os/pull/7453

AGlass0fMilk, 2018-07-09:
I synced my fork (previously at 5.9.0) to master (5.9.2+) and applied this PR - so far it appears to be working as expected. BLE is able to connect/pair and the UART works as well 👍

marcuschangarm, 2018-07-09:
Awesome! Thank you for trying it out!

closed 2018-07-12 by AGlass0fMilk