ARMmbed/mbed-os #7444: Bug in CANMessage constructor

OmerRosler, 2018-07-08:
CANMessage constructor gets parameters to initialize the class members. 
However the type of the parameters doesn't match the type of its members. For example the type of message ID is `unsigned int` however the constructor receives an `int` value. This causes implicit casting from `int` to `unsigned int` when the value is assigned to the member.

0xc0170, 2018-07-09:
Please leave the section about the issue type (it was removed from the issue template)

aashishc1988, 2018-08-21:
Indeed a bug. Filed the PR for it.

aashishc1988, 2018-09-25:
This issue is fixed. I do not have permissions to close this issue (for some reason). Someone else would have to do that.

SenRamakri, 2018-09-25:
Fixed by https://github.com/ARMmbed/mbed-os/pull/7848

closed 2018-09-25 by SenRamakri

embeddedteam103, 2018-12-16:
The `data` and `length` fields (as well as the docs) should also be unsigned as was hinted in the original bug report. This needs to be reopened :)
I will send a PR to fix this soon.