ARMmbed/mbed-os #7447: Mbed OS 5.9.2 regression with ARM CC compilers (Empty Execution region description for region RW_IRAM1)

JanneKiiskila, 2018-07-09:
### Description

Mbed Cloud Client example has stopped linking with Mbed OS 5.9.2, both with ARM CC v5.06 update 6 and ARM CC v6.10.

### Issue request type
[ ] Question  
[ ] Enhancement  
[x] Bug

You can verify this with publicy available [mbed-cloud-client-example](https://github.com/ARMmbed/mbed-cloud-client-example).

Any target will do, all of the Client team supported targets fail.

Example log below.

```
11:45:59 Product: ARM Compiler 5.06
11:45:59 Component: ARM Compiler 5.06 update 3 (build 300)
11:45:59 + COMPILER=ARM
...
11:47:48 [mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #62f8b922b420
...
11:49:08 Link: app_application
11:49:15 "/tmp/RJ1d4g", line 124 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
11:49:15 Error: L6200E: Symbol __user_perthread_libspace multiply defined (by BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.o and BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/mbed_boot.o).
11:49:15 Not enough information to list the image map.
11:49:15 Finished: 1 information, 1 warning and 1 error messages.
11:49:15 [ERROR] "/tmp/RJ1d4g", line 124 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1
11:49:15 Error: L6200E: Symbol __user_perthread_libspace multiply defined (by BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_lib.o and BUILD/K64F/ARM/mbed-os/rtos/TARGET_CORTEX/mbed_boot.o).
11:49:15 Not enough information to list the image map.
11:49:15 Finished: 1 information, 1 warning and 1 error messages.
11:49:15 
11:49:15 [mbed] ERROR: "/usr/bin/python" returned error code 1.
11:49:15 [mbed] ERROR: Command "/usr/bin/python -u /builds/ws/cloud-client-example-multistack/38b909a9/app/mbed-os/tools/make.py -t ARM -m K64F --profile profiles/debug_size.json --source . --build ./BUILD/K64F/ARM -DRESET_STORAGE -DRESET_FIRMWARE" in "/builds/ws/cloud-client-example-multistack/38b909a9/app"
11:49:15 ---
11:49:15 Build step 'Execute shell' marked build as failure
```

and same for ARMC 6.

```
12:21:47 + armclang --version
12:21:55 Product: ARM Compiler 6.10 Professional
12:21:55 Component: ARM Compiler 6.10
12:21:55 + COMPILER=ARMC6
...
12:25:08 |- mbed-os (#62f8b922b420, tags: latest, mbed-os-5.9.2)
...
12:28:45 Warning: L6439W: Multiply defined Global Symbol SingletonPtr<rtos::Mutex>::get() defined in invalid_group(FlashIAP.o) rejected in favor of Symbol defined in .text._ZN12SingletonPtrIN4rtos5MutexEE3getEv(nsapi_dns.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void(int)>::call(int) const defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFviEE4callEi(SerialBase.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol SimpleSpirit1::disable_spirit_irq() defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZN13SimpleSpirit118disable_spirit_irqEv(radio_spi.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol mbed::DirHandle::readdir()::ent defined in invalid_group(Dir.o) rejected in favor of Symbol defined in .bss._ZZN4mbed9DirHandle7readdirEvE3ent(FileSystem.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol SimpleSpirit1::enable_spirit_irq() defined in invalid_group(SimpleSpirit1.o) rejected in favor of Symbol defined in .text._ZN13SimpleSpirit117enable_spirit_irqEv(radio_spi.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol SingletonPtr<mbed::Timeout>::get() defined in invalid_group(utest_shim.o) rejected in favor of Symbol defined in .text._ZN12SingletonPtrIN4mbed7TimeoutEE3getEv(arm_hal_timer.o).
12:28:45 Warning: L6439W: Multiply defined Global Symbol mbed::Callback<void()>::call() const defined in invalid_group(ATParser.o) rejected in favor of Symbol defined in .text._ZNK4mbed8CallbackIFvvEE4callEv(SerialBase.o).
12:28:45 Error: L6200E: Symbol __user_perthread_libspace multiply defined (by rtx_lib.o and mbed_boot.o).
12:28:45 Not enough information to list the image map.
12:28:45 Finished: 1 information, 19 warning and 1 error messages.
12:28:45 [ERROR] Warning: L3912W: Option 'legacyalign' is deprecated.
12:28:45 Warning: L3912W: Option 'no_strict_wchar_size' is deprecated.
12:28:45 Warning: L3912W: Option 'no_strict_enum_size' is deprecated.
12:28:45 "/tmp/pjdwet", line 35 (column 3): Warning: L6312W: Empty Execution region description for region RW_IRAM1

```

teetak01, 2018-07-09:
Cloud Client example delivers its own profile-files for size-optimization and the 5.9.2 release broke the ARMCC/ARMC6 compilation with the debug_size profile. That is somewhat out-of-date and will be updated, although still curious that patch-release of Mbed OS caused this breakage.

0xc0170, 2018-07-09:
Based on the error, this is a duplicate of https://github.com/ARMmbed/mbed-os/issues/7422 ?

cc @theotherjimmy

JanneKiiskila, 2018-07-09:
Hash 76d7bb0d on 5.9 branch still links.

JanneKiiskila, 2018-07-09:
I believe hash 17203413dc6aaa4f7eec73abbc1a496967dbc718 is now the one breaking it.

0xc0170, 2018-07-09:
Thanks @JanneKiiskila . The macro somehow does not get into the rtx lib and the rtx armcc userspace for lib is still used. we will investigate

0xc0170, 2018-07-09:
I noticed the note above, requires own profile. That would explain the failures. RTE symbol is needed, it's part of the profiles. As the macro was placed correctly to the mbed config, it now requires RTE symbol unfortunately. 

This is duplicate of the above issue however the cause is own profile used. 

@ARMmbed/mbed-os-core @ARMmbed/mbed-os-tools  Please review this issue

theotherjimmy, 2018-07-09:
@0xc0170 We should not have put this in the profiles. It makes more sense in the `rtos/mbed_lib.json`.

0xc0170, 2018-07-09:
@theotherjimmy Can you fix it please? 5.9.3 would have it then fixed and backward compatible 💯

theotherjimmy, 2018-07-09:
See https://github.com/ARMmbed/mbed-os/pull/7449

theotherjimmy, 2018-09-24:
Yep. This is no longer an issue.

closed 2018-09-24 by theotherjimmy