ARMmbed/mbed-os #7450: Codeblocks disabled in online compiler

theotherjimmy, 2018-07-09:
### Description

The "codeblocks" exporter is disabled in the online compiler. This is because every single export with that exporter fails. The exporter tries to create a directory in the current working directory, which is not writable in the online compiler.

### Issue request type

[ ] Question  
[ ] Enhancement  
[x] Bug

0xc0170, 2018-07-10:
@ARMmbed/team-onsemi @tsailer Please review

tsailer, 2018-07-10:
AFAIK it only creates a 'bin' and 'obj' directory in the project top, and then puts an .mbedignore file into them.

I could skip creating those directories, as C::B creates them automatically if they don't exist. But the problem is then that after a C::B build, mbed builds don't work anymore because they try to link the C::B compiled objects in the obj and bin directories, thus linking stale object files that likey have symbol clashes. Hence creating them and putting .mbedignore into them.

Any idea what the best way forward is? I have zero knowledge about the online compiler

theotherjimmy, 2018-07-10:
@tsailer Thanks for the quick response. You would need to take the `self.export_dir` into account when creating those directories, or we could enhance `self.gen_file` to call makedir as appropriate and you could drop the mkdir calls.

theotherjimmy, 2018-07-10:
I'm a fan of the latter solution:
 * enhance `self.gen_file` to make directories when it needs to
 * drop the calls to `mkdir`.

tsailer, 2018-07-11:
@theotherjimmy this looks like a good solution to me too. Are you going to enhance self.gen_file? I can make use of this afterwards

theotherjimmy, 2018-07-12:
@tsailer Yep. I'm doing it right now. I'll include the PR in this issue when it's up.

theotherjimmy, 2018-07-16:
@tsailer The first half of the implementation, enhance `self.gen_file`, is done. Feel free to start working on making use of it.

maclobdell, 2018-08-16:
@tsailer @theotherjimmy  What is the status of this?  Is there still another PR to be made to get this working?

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTPART-5795*

adbridge, 2021-06-07:
As there has been no update on this for over 2 years I am closing it. Please re-open if this is still a valid issue.

closed 2021-06-07 by adbridge