ARMmbed/mbed-os #7451: SecurityManager Persistent Bonding Example

AGlass0fMilk, 2018-07-09:
### Description
Hi,
I'm trying to get persistent bonding working on an nRF52832 using external flash on mbed-os-5.9.0. 

I wrote the SPI flash driver to be compatible with mbed's BlockDevice API and LittleFS is able to properly mount a filesystem on it. I have given the SecurityManager.init call a filepath and based on my logic analyzer results it seems to be reading/writing to the external flash. However, the device fails to reconnect with a bonded device upon reset. Additionally, a call to generateWhitelistFromBondTable returns with a size of 0. So it would appear the bond table is not being saved to external flash properly?

The docs here: https://github.com/ARMmbed/mbed-os-5-docs/blob/development/docs/reference/api/connectivity/bluetooth/SecurityManager.md

say there are a few requirements: 
1.) Initialize a filesystem (I am using littlefs and mounting returns with no errors)
2.) Provide `SecurityManager.init` with a file path to save to
3.) Call `preserveBondingStateOnReset(true)`
with the caveat that 
>"Persistence may fail if abnormally terminated"

To ensure I'm not "abnormally terminating", I schedule a controlled reset 8 seconds after on_connect is called:
```void disconnect_and_reset(void)
{
	BLE& ble = BLE::Instance();
        // Disconnect, reset SM, shutdown BLE, and issue a system reset
	ble.gap().disconnect(Gap::LOCAL_HOST_TERMINATED_CONNECTION);
	ble.securityManager().reset();
	wait_ms(100);
	ble.shutdown();
	wait_ms(300);
	NVIC_SystemReset();
}
```
I'm not ruling out that my flash driver is to blame, but I would like a bit more clarification/an example of how to properly use the new persistent bonding features of the SM and how to tell if it has successfully saved information.

Thanks!

### Issue request type

[ X ] Question  
[ ] Enhancement  
[ ] Bug

Ashish285, 2018-07-26:
I am having a similar issue. 
I have created a Heap Block Device, and using Little File System. The filesystem is mounted properly and I can read and write to a file on it. But the security manager doesn't save anything and if i check the file, it is empty.

donatieng, 2018-08-08:
@paul-szczepanek-arm please review