ARMmbed/mbed-os #7455: init/deinit conflict among utility block devices on the same SD card

ccli8, 2018-07-10:
### Description

Utility block devices (`ChainingBlockDevice`/`SlicingBlockDevice`/`MBRBlockDevice`) are constructed upon e.g. `SDBlockDevice`. If multiple utility block devices are constructed on the same underlying block device, there would be `init`/`deinit` conflict among them. Take `SlicingBlockDevice` as an example: its `init`/`deinit` would call `init`/`deinit` of the underlying block device (`_bd`).
```
int SlicingBlockDevice::init()
{
    int err = _bd->init();
```
```
int SlicingBlockDevice::deinit()
{
    return _bd->deinit();
```

I can think of two issues with this:
1. `init` of the underlying block device would be called multiple times by utility block devices constructed on it.
1. Calling `deinit` of the underlying block device in one utility block device would fail another utility block device constructed on the same underlying block device.

### Issue request type

[ ] Question  
[ ] Enhancement  
[X] Bug  

### Related issue

https://github.com/ARMmbed/simple-mbed-cloud-client/issues/13

davidsaada, 2018-07-11:
This is indeed problematic. Had similar problems when using chained BDs or when BD was included as a member of class.
Maybe we should have an init/deinit reference count. @geky What do you think?

geky, 2018-07-12:
Agreed. I've noticed this problem for the SlicingBlockDevice/MBRBlockDevice since they share a single resource, but is it an issue for the ChainingBlockDevice? I don't think we end up with multiple ChainingBlockDevice's using one resource.

Reference counting init/deinit would solve the problem, but unfortunately it would need to be in the block device drivers.

Note that a lot of the block device drivers are stateless and calling init multiple times has no side effects. I believe spif, dataflash, and i2ceeprom all fall into this category.

cc @deepikabhavnani

davidsaada, 2018-07-12:
Having an `initialized` variable for the block devices should solve the multiple init problem. As for deinit - maybe we should not call it automatically from the calling block device.

davidsaada, 2018-07-17:
OK, will handle it.

davidsaada, 2018-07-30:
> Yes reference count along with existing init flag _is_initialized in SD driver can resolve this.

@deepikabhavnani Why would you need two variables for it? A value of 0 in the ref count means that the device is uninitialized, while all other values means it is initialized. In this case, the `_is_initialized` is redundant.

deepikabhavnani, 2018-07-30:
@davidsaada - Yes we can use reference count to check the device initialization status as well.. I just wanted to point that we have `_is_initialized` functionality as well which should be take care of.

davidsaada, 2018-07-30:
Implemented  [#7468](https://github.com/ARMmbed/mbed-os/pull/7648) for all internal mbed-os BDs. Will handle the external ones (SD, SPIF etc.) once this PR is merged (as the init_ref_count variable has been added to the base class).

davidsaada, 2018-08-01:
@0xc0170 Maybe this was prematurely closed. I still need to implement this on the SD block device (and other external BDs such as SPIF). [#7468](https://github.com/ARMmbed/mbed-os/pull/7648) only implements it for the internal mbed-os block devices.

0xc0170, 2018-08-01:
It was automatically. Reopening

davidsaada, 2018-08-02:
Ugh... Needs to be reopened again (At least SPIF should include this fix).

geky, 2018-08-02:
@davidsaada, just in case you didn't know, the "resolves" keyword in your PRs is what's closing the issue:
https://help.github.com/articles/closing-issues-using-keywords

I don't know if you want this reopened again.

closed 2018-08-02 by geky

davidsaada, 2018-08-02:
>just in case you didn't know, the "resolves" keyword in your PRs is what's closing the issue

Thanks, learning something new every day.

davidsaada, 2018-08-02:
And don't think I need it reopened. Do you think I need to do that in other external BDs (namely DataFlash and FlashIAP)?

MarceloSalazar, 2018-08-15:
@geky can you please confirm @davidsaada question?

davidsaada, 2018-08-15:
@MarceloSalazar That won't be necessary. Did that in all left external BDs (and this was reviewed by @geky as well).

MarceloSalazar, 2018-08-15:
Thanks for confirming @davidsaada !