ARMmbed/mbed-os #7456: Compilation duration optimization required

adustm, 2018-07-10:
### Description
Dear mbed-os core team
Time goes by, the amount of features is magically increasing. All this is awesome, for sure.
But mbed-os takes me more than 10 minutes to compile completely. 

Could you spend some time to cleanup the compilation duration ? Or give me advices to avoid full recompilation ?
Here are some ideas : 
* Make sure I don't compile features that are not required for the platform I am compiling
(cellular feature is not required on the DISCO_L475VG_IOT01A for instance)
* improve mbed_app.json mechanism so that it does not compile every files every time
(let's say I am only changing the SSID and PASSWORD of the wifi, it is used in 1 or 2 cpp files, but the full mbed-os database is recompile

### Issue request type
    [] Question
    [X] Enhancement 
    [ ] Bug

adustm, 2018-07-10:
Thanks for the advice @mstfkaratas 
The problem with this .mbedignore is that not everyone has the exact knowledge of which feature he/she needs for any kind of application. At the end, if your application does not work, you will spend time in debug for something that is natively working. 

I understand that it is very difficult for mbed-os team to know what to remove or not.

kind regards

jeromecoutant, 2018-07-10:
Build script is able to parse targets directory to compile only the correct target.
So it should be able to parse features directory depending on DEVICE_HAS macro ?

jeromecoutant, 2018-09-04:
@theotherjimmy 

> improve mbed_app.json mechanism so that it does not compile every files every time
> (let's say I am only changing the SSID and PASSWORD of the wifi, it is used in 1 or 2 cpp files, but the > full mbed-os database is recompile).

Do you think this is feasible to detect only the change in the json files ?

Thx

adbridge, 2018-10-04:
*Internal Jira reference: https://jira.arm.com/browse/IOTCORE-245*

closed 2019-10-31 by ciarmcom